fr3ts0n/AndrOBD #1: AWT/Swing for Android?

krt16s, 2015-05-24:
It looks like you are using AWT and swing components and compiling it gives me package warnings. Is there some sort of AWT/Swing-to-Android library that I am missing?

fr3ts0n, 2015-05-24:
There are some swing / awt files in the repo, since the repo also hosts swing-based OBD- and diagnostic applications, but these are excluded from compile in my projects. (Eclipse and Android Studio)
What build environment do you use?

krt16s, 2015-05-25:
The buildserver doesn't have eclipse nor android studio, but uses gradle (for AS builds) and ant (for Eclipse builds) directly. Which files should be excluded?

fr3ts0n, 2015-05-25:
`<classpathentry excluding="com/fr3ts0n/pvs/gui/|com/fr3ts0n/ecu/gui/application/|com/fr3ts0n/ecu/prot/Kw1281Prot.java|com/fr3ts0n/prot/gui/" kind="src" path="src"/>`

krt16s, 2015-05-25:
Thanks for pointing out, builds by removing them.

closed 2015-05-25 by krt16s