fr3ts0n/AndrOBD #2: usb support ?

xantares, 2015-05-30:
hello,
from what I understand from the fdroid doc it only supports bluetooth devices,
what would it take to handble an usb eml327 device ?
regards

fr3ts0n, 2015-05-30:
It would just take another communication handler for the USB device, but I am not sure if all Android kernels provide support to connect the ELM adapter as a serial device.
I haven't tried this yet, since the handling of a bluetooth device with a mobile is so much easier (and the ELM BT adapters are really low cost)

rp3tya, 2015-09-24:
The USB API is officially supported in Android 3.1 and higher, it is easier and safer than bluetooth.
I have the hardware to test, but no time to implement a new handler.
Cheers

fr3ts0n, 2015-10-09:
USB support implemented with V1.3.0

closed 2015-10-09 by fr3ts0n

xantares, 2015-10-09:
awesome!