fr3ts0n/AndrOBD #4: Extended PIDs per Manufacturer

pfuender, 2015-06-03:
Hi there!

I've just got myself an ODB-Bluetooth adapter and wanted to play with some of the vendor specific PID's. Do you have any plan on making the app a bit more flexible in these terms? XML per vendor etc? Import/export or even directly in the app?
Additionally a scan function for an addressrange would also be great for reversing, so you can quickly compare two scans and find out the matching addresses for a sensor you like.

Cheers

fr3ts0n, 2015-06-03:
Hi,
I will split your request into two new issues:

1. Enhancement: Add some flexible way to handle additional / vendor specific PIDs
The current data structure does not allow to handle PIDs which are unknown. It just logs a warning message with PID number in the log file. Please could you send/attach the log file of your attempts that I can see what PID's it is all about? I may be able to implement these short term.
However, as mid-/long term solution I am planning change this handling to a more dynamic list which can be enhanced by loading .csv tables in addition to the pre-set.

2. Enhancement: Add a scan function for an addressrange
I think I need some more details on that one. 
Are you asking for scanning a range of ECU addresses? If so, we are probably talking about UDS protocol using vendor specific PID maps and conversions rather than OBD, which is standardized.
I know that it is possible with a ELM, but I'm just not there yet.

fr3ts0n, 2015-06-03:
Please respond on issues #5 and #6 
I will close this one

closed 2015-06-03 by fr3ts0n