fr3ts0n/AndrOBD #6: Add a scan function for an addressrange

fr3ts0n, 2015-06-03:
Enhancement: Add a scan function for an addressrange

I think I need some more details on that one.
Are you asking for scanning a range of ECU addresses? If so, we are probably talking about UDS protocol using vendor specific PID maps and conversions rather than OBD, which is standardized.
I know that it is possible with a ELM, but I'm just not there yet.

pfuender, 2015-06-12:
Well to be honest, I don't know much about the whole obd/udp/pid protocol stuff behind, but I was running [Torque](https://play.google.com/store/apps/details?id=org.prowl.torquescan), which has such a functionality. I guess once *you* have a look at it, you know what I'm talking about.

Thanks

fr3ts0n, 2015-10-26:
Allow ECU selection by address if more than one ECU respond to OBD requests.
implemented with V1.3.3

closed 2015-10-26 by fr3ts0n