fr3ts0n/AndrOBD #8: Blank screen: app says it's connected, logs says it isn't

JeroenVanOort, 2015-09-21:
I've found AndrOBD on F-droid and I'm looking forward to using it. But I just can't get it to work. I started the app, connected to the ELM327 in my (running) car but the screen stays empty. And although the indicator on the upper left corner tells me the app is connected to the ELM, I've found the log files to state else:

```
2015-09-21 19:34:34,604 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread started
2015-09-21 19:34:34,642 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'ELM327v2.1' (ATZ)
2015-09-21 19:34:34,643 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 1000 -> 12
2015-09-21 19:34:34,644 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: null->Initializing
2015-09-21 19:34:39,636 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 0->9
2015-09-21 19:34:39,692 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Initializing->Connecting
2015-09-21 19:34:39,693 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'SEARCHING...' (0900)
2015-09-21 19:34:40,197 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'UNABLETOCONNECT' (0900)
2015-09-21 19:34:40,198 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Connecting->Disconnected
```

I have found an app that does work (https://github.com/pires/android-obd-reader), but it doesn't have the features of your app. Can you please help me find our what's going wrong?

DreitTheDragon, 2015-09-22:
Exactly same issue here with both 1.2.6 and 1.2.7, didn't tried older versions and didn't checked log.
I'm using white ELM327 v2.1 dongle from AliExpress ($3), MAC address is simply 00:00:00:00:00:01 and it works with "Android OBD-II Reader" with no problem. I tried AndrOBD on three phones with no success.
Interesting thing - it seems to do something and fail immediately, because when I first time click for "OBD Vehicle Info" and then "OBD Data", some text blinks on screen. I'll try to catch it with camera, maybe tomorrow.

JeroenVanOort, 2015-09-22:
I've been playing with an ELM327 console app in the meanwhile and I have gotten the UNABLETOCONNECT message but only when not telling the module to automagically figure out the protocol used by the car (`ATSP0`). Is this command being send by AndrOBD? I can't seem to find it in the source code, but I'm only a humble PHP developer that has only seen flashes of Java in university.

fr3ts0n, 2015-09-22:
Hello Jeroen,
Thank you for your good description.
Yes,, AndrOBD sends a sp0 request, and I think this is a timing issue with the protocol search in the ELM.
I will geht back to you later on today.

fr3ts0n, 2015-09-22:
* It's a combination of setting automatic protocol detection (ATSP0) and setting a rather short response timeout (ATSTxx). Since all my testing cars appear to support CAN protocol on a rather fast rate, this hasn't been an issue yet.
* The status display (within the toolbar) currently only shows the bluetooth connection status.
* I will change it to represent the ELM communication status, as soon as we are in bluetooth ONLINE mode.

fr3ts0n, 2015-09-22:
Please verify correct behaviour of V1.2.8 on your vehicles, which will be available on F-Droid soon.

**Please note:**
_From V1.2.7 -> 1.2.8 the assignment of settings "Data items to display" to PIDs has changed.
If you see an empty data list with status connected, please clear application data (Settings -> Apps -> AndrOBD -> Clear data) and re-try_

fr3ts0n, 2015-09-28:
Do you have any results with V1.2.8 already?
Thx

JeroenVanOort, 2015-09-29:
I'm sorry to say it doesn't work:

```
2015-09-28 16:39:08,242 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread started
2015-09-28 16:39:08,389 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'ELM327v2.1' (ATZ)
2015-09-28 16:39:08,390 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 1000 -> 12
2015-09-28 16:39:08,391 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: null->Initializing
2015-09-28 16:41:46,745 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 0->9
2015-09-28 16:41:46,810 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Initializing->Connecting
2015-09-28 16:41:46,810 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'SEARCHING...' (0900)
2015-09-28 16:41:47,270 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'UNABLETOCONNECT' (0900)
2015-09-28 16:41:47,270 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Connecting->Disconnected
2015-09-28 16:41:47,271 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 12 -> 200
2015-09-28 16:41:47,310 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATSP0)
2015-09-28 16:41:47,368 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-28 16:41:47,404 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-28 16:41:47,442 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-28 16:41:47,477 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-28 16:41:47,513 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATST32)
2015-09-28 16:41:52,524 - [WARN::stream::com.fr3ts0n.prot.StreamHandler] - RX error
java.io.IOException: bt socket closed, read return: -1
	at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java:434)
	at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java:60)
	at com.fr3ts0n.prot.StreamHandler.run(StreamHandler.java:116)
	at com.fr3ts0n.ecu.gui.androbd.ObdCommService$ConnectedThread.run(ObdCommService.java:434)
2015-09-28 16:41:52,527 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread stopped
```

DreitTheDragon, 2015-09-29:
Same here. First time I deleted app data, started new AndrOBD in demo mode, closed app, opened app again, connected via bluetooth to ELM327 (at second try usually) and checked data. Doesn't matter what I was doing, there still were same values as when I was leaving demo mode + it wasn't possible to show any value in graph.

fr3ts0n, 2015-09-29:
Based on your log output the app tried to recover the protocol and timing correctly. 
The only thing which I would expect is a re-issue of the 0900 command, which I can see on my tests. Does the same thing happen on any re-try of this equence?
Please could you re-try and change to i.e. "OBD data" service and back to "OBD vehicle info", if you don't see more activity on the adapter.
Thx

fr3ts0n, 2015-09-29:
created V1.2.9 for testing/confirmation

issue #8 changes:

    * remove setting MIN timeout on service select (possible protocol search)
    * default timeout setting 200ms (dynamic adaption within loop)
    * make preferred OBD protocol configurable
    * init with echo off (eliminate multiple ate0's)

JeroenVanOort, 2015-09-30:
```
2015-09-30 16:41:53,719 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread started
2015-09-30 16:41:53,859 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'ELM327v2.1' (ATZ)
2015-09-30 16:41:53,860 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 1000 -> 12
2015-09-30 16:41:53,860 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: null->Initializing
2015-09-30 16:42:05,890 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 0->9
2015-09-30 16:42:05,946 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Initializing->Connecting
2015-09-30 16:42:05,947 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'SEARCHING...' (0900)
2015-09-30 16:42:06,420 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'UNABLETOCONNECT' (0900)
2015-09-30 16:42:06,426 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Connecting->Disconnected
2015-09-30 16:42:06,435 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 12 -> 200
2015-09-30 16:42:06,476 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATSP0)
2015-09-30 16:42:06,512 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-30 16:42:06,548 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-30 16:42:06,586 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-30 16:42:06,622 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-30 16:42:06,660 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATST32)
2015-09-30 16:42:08,382 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 9->1
2015-09-30 16:42:08,383 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 200 -> 12
2015-09-30 16:42:11,672 - [WARN::stream::com.fr3ts0n.prot.StreamHandler] - RX error
java.io.IOException: bt socket closed, read return: -1
	at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java:434)
	at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java:60)
	at com.fr3ts0n.prot.StreamHandler.run(StreamHandler.java:116)
	at com.fr3ts0n.ecu.gui.androbd.ObdCommService$ConnectedThread.run(ObdCommService.java:434)
2015-09-30 16:42:11,678 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread stopped
2015-09-30 16:42:13,406 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 1->3
2015-09-30 16:42:13,408 - [WARN::stream::com.fr3ts0n.prot.StreamHandler] - TX error:'30 33  : 03':java.io.IOException: socket closed
2015-09-30 16:42:18,254 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread started
2015-09-30 16:42:18,291 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'ELM327v2.1' (ATZ)
2015-09-30 16:42:18,291 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 1000 -> 12
2015-09-30 16:42:18,292 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Disconnected->Initializing
2015-09-30 16:42:30,365 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 3->9
2015-09-30 16:42:30,413 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Initializing->Connecting
2015-09-30 16:42:30,414 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'SEARCHING...' (0900)
2015-09-30 16:42:30,890 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'UNABLETOCONNECT' (0900)
2015-09-30 16:42:30,891 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Connecting->Disconnected
2015-09-30 16:42:30,892 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 12 -> 200
2015-09-30 16:42:30,930 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATSP0)
2015-09-30 16:42:30,967 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-30 16:42:31,003 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-30 16:42:31,041 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-09-30 16:42:31,077 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATST32)
2015-09-30 16:42:34,187 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 9->1
2015-09-30 16:42:34,188 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 200 -> 12
2015-09-30 16:42:36,089 - [WARN::stream::com.fr3ts0n.prot.StreamHandler] - RX error
java.io.IOException: bt socket closed, read return: -1
	at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java:434)
	at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java:60)
	at com.fr3ts0n.prot.StreamHandler.run(StreamHandler.java:116)
	at com.fr3ts0n.ecu.gui.androbd.ObdCommService$ConnectedThread.run(ObdCommService.java:434)
2015-09-30 16:42:36,090 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread stopped
```

This is my log file after trying again.

fr3ts0n, 2015-10-03:
Did anyone test this with 1.2.9 already?

JeroenVanOort, 2015-10-04:
I'm sorry to say 1.2.9 doesn't work for me either:

```
2015-10-04 12:24:56,491 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread started
2015-10-04 12:24:56,570 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: null->Initializing
2015-10-04 12:24:56,571 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 1000 -> 200
2015-10-04 12:24:56,702 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-10-04 12:24:56,836 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATL0)
2015-10-04 12:24:56,974 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATS0)
2015-10-04 12:24:57,109 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATST32)
2015-10-04 12:24:57,245 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATSP0)
2015-10-04 12:25:09,899 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 0->9
2015-10-04 12:25:14,920 - [WARN::stream::com.fr3ts0n.prot.StreamHandler] - RX error
java.io.IOException: bt socket closed, read return: -1
	at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java:434)
	at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java:60)
	at com.fr3ts0n.prot.StreamHandler.run(StreamHandler.java:116)
	at com.fr3ts0n.ecu.gui.androbd.ObdCommService$ConnectedThread.run(ObdCommService.java:434)
2015-10-04 12:25:14,922 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread stopped
2015-10-04 12:25:18,938 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread started
2015-10-04 12:25:18,975 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 1000 -> 200
2015-10-04 12:25:19,011 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-10-04 12:25:19,049 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATL0)
2015-10-04 12:25:19,084 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATS0)
2015-10-04 12:25:19,119 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATST32)
2015-10-04 12:25:19,159 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATSP0)
2015-10-04 12:25:19,161 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 200 -> 196
2015-10-04 12:25:24,140 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 9->1
2015-10-04 12:25:24,173 - [WARN::stream::com.fr3ts0n.prot.StreamHandler] - RX error
java.io.IOException: bt socket closed, read return: -1
	at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java:434)
	at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java:60)
	at com.fr3ts0n.prot.StreamHandler.run(StreamHandler.java:116)
	at com.fr3ts0n.ecu.gui.androbd.ObdCommService$ConnectedThread.run(ObdCommService.java:434)
2015-10-04 12:25:24,175 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread stopped
2015-10-04 12:25:28,480 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread started
2015-10-04 12:25:28,509 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 1000 -> 200
2015-10-04 12:25:28,549 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-10-04 12:25:28,587 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATL0)
2015-10-04 12:25:28,623 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATS0)
2015-10-04 12:25:28,659 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATST32)
2015-10-04 12:25:28,704 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATSP0)
2015-10-04 12:25:28,741 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATST31)
2015-10-04 12:25:28,742 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 200 -> 196
2015-10-04 12:25:31,042 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 1->9
2015-10-04 12:25:33,752 - [WARN::stream::com.fr3ts0n.prot.StreamHandler] - RX error
java.io.IOException: bt socket closed, read return: -1
	at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java:434)
	at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java:60)
	at com.fr3ts0n.prot.StreamHandler.run(StreamHandler.java:116)
	at com.fr3ts0n.ecu.gui.androbd.ObdCommService$ConnectedThread.run(ObdCommService.java:434)
2015-10-04 12:25:33,753 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread stopped
```

What can I do to help?

fr3ts0n, 2015-10-05:
Could you please describe your test environment?
* adapter type
* mobile device 
  + android version
* vehicle type
  + used protocol type

It appears that you always loose the bluetooth connection to the adaptor as soon as OBD connection gets established.
Could you please re-test this with logging depth DEBUG and attach that complete log file (as a file)? That should show us all communication i/o.

You also may want to select your default protocol type to the one your vehicle supports.

If you connect your device with a BT terminal app and repeat following command sequence, do you also loose bluetooth connection?
>ATZ
>ATE0
>ATL0
>ATS0
>ATST32
>ATSP0
>ATST31
>0900

Thx

JeroenVanOort, 2015-10-07:
I tried setting the protocol and that gives me the parameter list, but they are all zero-ish. I'm starting to doubt that the ELM327 is working correctly; it seems to send NODATA all the time and the other app didn't work flawless either. I'll try with a car that uses another protocol this weekend.

fr3ts0n, 2015-10-09:
I have also noticed some issues with a USB ELM327v2.1 which responded quite different than my BT ELM327v1.5. 
I have fixed these issues with V1.3.0 which should be available on F-Droid soon. 
(it now also supports USB devices)

JeroenVanOort, 2015-10-14:
I just installed the update and I'll try to test it tonight.

fr3ts0n, 2015-10-20:
Any results?

JeroenVanOort, 2015-10-26:
I've tested once more with my first dongle and since this morning I've been testing with a new one, because I thought the first one was dodgy. It turns out the second one gives me the same bt socket closed messages. Meanwhile I've found out that custom ROM's don't always play nicely with bluetooth connections on Nexus 5 devices.

I'll try to work around that.

fr3ts0n, 2015-11-07:
With V1.3.5 I have changed the default bluetooth connection to a plain SPP instead of encrypted.
* You can give it a try...

Encrypted SPP can still be configured in settings.

If your car is using KWP protocol you may also want to add some custom initialisation commands (except selecting protocol 5 as your default) Settings -> OBD Settings -> Custom ELM init
>ATSH8010F1
>ATWM8010F00105
>ATSW10

fr3ts0n, 2016-02-05:
I'm going to close this now.
If this is still an issue, then please re-open, or create new.

closed 2016-02-05 by fr3ts0n