fr3ts0n/AndrOBD #9: logging doesn't seem to work

kavol, 2015-10-08:
The program doesn't work with my car (while e.g. OBD Car Doctor does, so it is not an adapter/protocol issue). So I wanted to take a look what may be wrong, if right protocol is used etc.

I went to settings => Development options => Logging level

This is set to "DEBUG".

However, it doesn't write anything onto the screen. And trying to find any logfile, I've failed too. There is nothing in /sdcard/com.fr3ts0n.ecu.gui.androbd/ and I was unable to find anything relevant trying to search for something with 'androbd' in name, or 'jobd.log'.

fr3ts0n, 2015-10-08:
The log files are located in com.fr3ts0n.ecu.gui.androbd**/log/** on your mobile's primary storage media.  The files are called **AndrOBD.log, AndrOBD.log.1 ...**

All log enties may also be recorded by connecting a logcat reader like adb via USB.

Could you please explain the symptoms of "doesn't work with my car"?

kavol, 2015-10-15:
Ok, I got the file now, exactly in that location that I've commented "There is nothing in /sdcard/com.fr3ts0n.ecu.gui.androbd/" (=> log/AndrOBD.log). Looks like it magically started working just after I've reported this, as the first record has the same date. But I've played with this app before and there are no older log entries ...

So I guess this could be closed as working in the recent version.

As for "doesn't work with my car", I'd rather open a new ticket attaching the logfile, not to change topic of the issue. In short. it establishes the bluetooth connection, but then it does nothing, is says "Connecting" foever, but the main screen remains empty, it shows nothing trying to display OBD Vehicle Info or anything else.

closed 2015-10-15 by kavol