fr3ts0n/AndrOBD #10: can't connect - Lancia Phedra 2.2 JTD (4HW)

kavol, 2015-10-15:
Trying to connect to my car, the bluetooth connection is established, but it seems that the application cannot talk to the ECU. It just writes "Connecting" forever, and the main screen is empty, no OBD Vehicle Info shown, or anything else.

Trying with OBD Car Doctor, it says it uses protocol 5 - KWP fast.

(Sorry for pasting the log instead of attaching it, but I got "Attaching documents requires write permission to this repository.  Try again with a PNG, GIF, or JPG.")

2015-10-08 14:01:43,367 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM DEMO thread started
2015-10-08 14:02:44,272 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Preferred protocol: Automatic
2015-10-08 17:02:15,675 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM DEMO thread finished
2015-10-15 08:01:13,340 - [DEBUG::CommService::com.fr3ts0n.ecu.gui.androbd.BtCommService] - connect to: 00:0D:18:00:00:01
2015-10-15 08:01:13,495 - [DEBUG::CommService::com.fr3ts0n.ecu.gui.androbd.CommService] - setState() NONE -> CONNECTING
2015-10-15 08:01:13,742 - [INFO::CommService::com.fr3ts0n.ecu.gui.androbd.BtCommService$BtConnectThread] - BEGIN mBtConnectThread SocketType:Insecure
2015-10-15 08:01:15,871 - [DEBUG::CommService::com.fr3ts0n.ecu.gui.androbd.BtCommService] - connected, Socket Type:Insecure
2015-10-15 08:01:15,876 - [DEBUG::CommService::com.fr3ts0n.ecu.gui.androbd.CommService] - setState() CONNECTING -> CONNECTED
2015-10-15 08:01:15,892 - [DEBUG::CommService::com.fr3ts0n.ecu.gui.androbd.BtCommService$BtWorkerThread] - create BtWorkerThread: Insecure
2015-10-15 08:01:15,905 - [INFO::CommService::com.fr3ts0n.ecu.gui.androbd.BtCommService$BtWorkerThread] - BEGIN mBtWorkerThread
2015-10-15 08:01:15,909 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread started
2015-10-15 08:01:15,919 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:'ATZ'
2015-10-15 08:01:15,924 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.prot.TelegramSender] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:41 54 5A  : ATZ
2015-10-15 08:01:16,767 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'ELM327v1.5'
2015-10-15 08:01:16,773 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: null->Initializing
2015-10-15 08:01:16,778 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM learn timeout: 12 -> 12
2015-10-15 08:01:16,787 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM Timeout: 1000 -> 200
2015-10-15 08:01:16,790 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'>'
2015-10-15 08:01:16,798 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:'ATE0'
2015-10-15 08:01:16,804 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.prot.TelegramSender] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:41 54 45 30  : ATE0
2015-10-15 08:01:16,848 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'ATE0'
2015-10-15 08:01:16,858 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'OK'
2015-10-15 08:01:16,860 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATE0)
2015-10-15 08:01:16,870 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'>'
2015-10-15 08:01:16,871 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:'ATL0'
2015-10-15 08:01:16,877 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.prot.TelegramSender] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:41 54 4C 30  : ATL0
2015-10-15 08:01:16,916 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'OK'
2015-10-15 08:01:16,919 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATL0)
2015-10-15 08:01:16,921 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'>'
2015-10-15 08:01:16,923 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:'ATS0'
2015-10-15 08:01:16,925 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.prot.TelegramSender] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:41 54 53 30  : ATS0
2015-10-15 08:01:16,951 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'OK'
2015-10-15 08:01:16,953 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATS0)
2015-10-15 08:01:16,954 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'>'
2015-10-15 08:01:16,955 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:'ATST32'
2015-10-15 08:01:16,958 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.prot.TelegramSender] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:41 54 53 54 33 32  : ATST32
2015-10-15 08:01:16,990 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'OK'
2015-10-15 08:01:16,991 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATST32)
2015-10-15 08:01:16,993 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'>'
2015-10-15 08:01:16,994 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:'ATSP0'
2015-10-15 08:01:16,997 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.prot.TelegramSender] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:41 54 53 50 30  : ATSP0
2015-10-15 08:01:17,023 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'OK'
2015-10-15 08:01:17,024 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATSP0)
2015-10-15 08:01:17,025 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'>'
2015-10-15 08:01:17,026 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Initializing->Initialized
2015-10-15 08:01:30,330 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - OBD Service: 0->9
2015-10-15 08:01:30,337 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:'0900'
2015-10-15 08:01:30,339 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.prot.TelegramSender] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:30 39 30 30  : 0900
2015-10-15 08:01:30,389 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'SEARCHING...'
2015-10-15 08:01:30,390 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Initialized->Connecting
2015-10-15 08:01:30,393 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'SEARCHING...' (0900)
2015-10-15 08:01:35,862 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'UNABLETOCONNECT'
2015-10-15 08:01:35,864 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'UNABLETOCONNECT' (0900)
2015-10-15 08:01:35,865 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'>'
2015-10-15 08:01:35,866 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - Status change: Connecting->Disconnected
2015-10-15 08:01:35,869 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:'ATSP0'
2015-10-15 08:01:35,879 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.prot.TelegramSender] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:41 54 53 50 30  : ATSP0
2015-10-15 08:01:35,917 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'OK'
2015-10-15 08:01:35,918 - [INFO::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - ELM rx:'OK' (ATSP0)
2015-10-15 08:01:35,921 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'>'
2015-10-15 08:01:35,922 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:'0900'
2015-10-15 08:01:35,932 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.prot.TelegramSender] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 TX:30 39 30 30  : 0900
2015-10-15 08:01:35,987 - [DEBUG::com.fr3ts0n.prot::com.fr3ts0n.ecu.prot.ElmProt] - com.fr3ts0n.ecu.prot.ElmProt@41e4f3b0 RX:'SEARCHING...'
[... cut ...]
	at com.fr3ts0n.ecu.gui.androbd.BtCommService$BtWorkerThread.run(BtCommService.java:345)
2015-10-15 08:06:04,574 - [INFO::stream::com.fr3ts0n.prot.StreamHandler] - RX Thread stopped
2015-10-15 08:06:04,576 - [DEBUG::CommService::com.fr3ts0n.ecu.gui.androbd.BtCommService] - stop
2015-10-15 08:06:04,608 - [DEBUG::CommService::com.fr3ts0n.ecu.gui.androbd.CommService] - setState() OFFLINE -> OFFLINE
2015-10-15 08:06:04,697 - [DEBUG::CommService::com.fr3ts0n.ecu.gui.androbd.CommService] - setState() OFFLINE -> OFFLINE

fr3ts0n, 2015-10-15:
Sorry to ask the stupid question, but:
Was the ignition key on?

The command/response sequence shows exactly the behaviour as if the key is off, or the adapter does not have a physical connection to the communication pins.

kavol, 2015-10-15:
Yes; not only the key was on but also the engine was running (the car behaves differently, so just in case ...)

fr3ts0n, 2015-10-15:
1st attempt could be to set the preferred protocol to "ISO 14230-4 KWP (fast Init)"
(Settings -> OBD Settings -> Preferred Protocol). Then re-start and re-connect again..

If that doesn't work we will have to find out, which command in the sequence prevents your car from answering:
Please try to send following sequence to your adapter via a bluetooth terminal? 
(Each line at a time and wait for the Prompt '>' again)
**ATZ** (Reset Adapter)
**ATE0** (Echo OFF)
**ATL0** (Line feeds OFF)
**ATS0** (Spaces OFF)
**ATST32** (Set response timeout to 200ms)
**ATSP0** (Set Protocol to automatic detection)
**0900** (get supported OBD service 9 PIDs)

Does it still refuse connection?
If it still does, try the same sequence without the ATST32 line.

Thx

fr3ts0n, 2015-10-20:
Any results?

kavol, 2015-10-20:
setting the protocol doesn't help

trying Bluetooth Terminal (the version available from F-Droid repo), I'm getting disconnected after 6-7 seconds ... and I mean disconnected from the adapter, not ECU

how to debug that, is there any easy tool to sniff BT traffic so that we can see what happens and compare to an application that actually works?

my whole testing session today looked like:
[start the car]
1) try to set the protocol in AndrOBD, did not help, same behaviour
2) run OBD Car Doctor to verify that the adapter is working ... okay, I can display e.g. RPM
3) run bluetooth terminal, choose the device, write ATZ, send, write ATE0, send ... send ...send? - why I can't click that? - ah, disconnected ... choose the device again, connected, try to write ATZ ... whoops, disconnected again?
4) run OBD Car Doctor again ... okay, I can display e.g. Battery Voltage
5) run bluetootth terminal again, choose the device, count ... about six seconnds and the connection is gone again ... try again, count ... connection lost again ... reconnect, write ATZ, reconnect, write ATE0, reconnect, write ATL0 etc. ... doesn't seem to lead anywhere, give up

kavol, 2015-10-23:
cool, I've found that Android itself allows to log BT traffic so here are the captures:

http://uloz.to/xNGXioow/androbd-btsnoop-hci-log
http://uloz.to/xPnNjttc/obdcardoctor-btsnoop-hci-log

(still getting errors trying to attach files to the issue)

looks like there are some differences in the initialization sequence, then AndrOBD gets error while OBD Car Doctor receives data ...

I haven't figured what's wrong with BT Terminal though; maybe I'll try to set line end to \r like both OBD programs use and then try the capture again, when I get bored over the weekend :-)

fr3ts0n, 2015-10-24:
Thanks for the captures.
I have found nothing exciting happening in the obdcardoctor log.
It only sets some protocol headers, which should not really be necessary for OBD access.
* Did you have to configure this in the app settings?

I still doubt if the KWP protocol can handle my setting of the timing parameters.
Therefore I have made a setting to disable the adaptive timing handling in the ELM protocol driver. 
-> V1.3.2 (should be availabe on F-Droid soon ...)

There is also a new development setting: "Disable ELM command" which can be used to identify ELM commands which cause problems on specific adapters.

kavol, 2015-11-02:
No, I haven't configured anything special in OBD Car Doctor, only the protocol not to wait for detection each time.

The new setting did not help. Unfortunately, I've misread that at the first try, and when I've realized the mistake and retried, I haven't retrieved the packet capture :-( However, the application behavior is the same, when trying to display vehicle info, it just disconnects over and over.

I've also tried to capture communication of EOBD Facile, which somehow works too:
http://uloz.to/xhVA8DtR/eobdfacile-btsnoop-hci-log

fr3ts0n, 2015-11-07:
With V1.3.5 I have changed the default bluetooth connection to a plain SPP instead of encrypted.
* You can give it a try...

Encrypted SPP can still be configured in settings.

You may also want to add some custom initialisation commands (except selecting protocol 5 as your default) Settings -> OBD Settings -> Custom ELM init
>ATSH8010F1
>ATWM8010F00105
>ATSW10

kavol, 2015-11-29:
Sorry for slow response. Priorities, priorities ...

Now we're getting somewhere - after using the custom sequence, it at least says that it is searching for ECU before disconnecting.

New logs here:
http://uloz.to/xSpodEu4/androbd-zip

First I've tried new version 1.3.7 with the old settings. It did disconnect immediately.
Then I've added the init sequence in the settings and restarted the app. After that it got to the "Searching for ECU" (or "Connecting to"? - don't remember) step.
At last I've run the OBD Car Doctor (displaying temperature reading) to verify the link works and the adapter is not stuck.

I have a suspicion that the bluetooth log contains only the communication of the OBD Car Doctor and not AndrOBD, but I don't understand BT communication enough to be sure ... but if it is so, I don't know how did it happen and how to capture the right thing - I didn't play with android settings since I've turned the logging on, I'm just deleting the old captures from time to time, but I always make sure to restart the phone after that to avoid stale file descriptors ...

fr3ts0n, 2015-11-30:
Good to hear that the bluetooth connection now stays stable ...

You are right, the hci log only contains the OBD car doctor communication, but the AndrOBD log helps for comparison.
I have compared the hci log with the AndrOBD an found the following differences:
* you seem to have selected protocol 4 (KWP 5-baud init) as your default protocol (in AndrOBD)
  * OBD car doctor uses protocol 5 as the default (KWP with fast init) 
  - => Please change your preferred protocol to 5 (KWP-fast init)
* OBD car doctor uses following initialisation commands
  - => you can enter these as custom initialisation commands (replacing the previous ones):

>ATIB10
>ATIIA10
>ATSH8110FC

That should do the job ...

kavol, 2015-12-09:
re protocol 4/5 - oops, stupid me, you're right, fat fingers ...

now it connects, but it doesn't receive data; new logs:

http://uloz.to/xMiBcgAC/androbd-2015-12-03-zip

kavol, 2015-12-09:
ok, I've found some more time today, so I've tried to change the init sequence to the previous one suggested, now that I know the protocol setting was wrong ... and it works, cool!

meanwhile, I've replaced MAF, so I've used that to check the new one reports correct values, screenshot attached to the new logs to show how the app works:

http://uloz.to/xeD2syQp/androbd-2015-12-09-zip

I haven't examined all the readings thoroughly, so maybe I have missed something, but ... what works seems to return correct values, what doesn't work is:
* vehicle info doesn't show anything
* throttle position stays at zero

so ... if you could, pretty please with sugar on top, implement some detection magic so that the parameters which make this work don't have to be entered manually, it would be awesome and this report could be closed then - many thanks for finding the solution!

fr3ts0n, 2015-12-10:
Ok, It seems we're getting there slowly, but steady.

### vehicle info
In the AndrOBD log I have seen several responses like:
> Service not (yet) supported: 63

after you selected service 9 (Vehicle information)

The log warning means that 1st byte of all the responses in mode 9 is 63 (0x3F) which can not be interpreted as a service response.
Please could you do the same test again with logging depth set to DEBUG? This will enable logging all data communication with the ELM adapter rather than just the control commands (in AndrOBD log). 
(Unfortunately the btsnoop log just contains service 1 data requests, but no service 9, or any other)

### throttle position
Your ECU really responded with 0000 to all throttle position requests ( 01 11 ) I could find in the btsnoop log!
There doesn't seem to be much I can do about that.

### custom init commands
What I am wondering about is your custom init commands:
> ATWM8010F00105

* This sets the network wakeup (keep alive) message which is usually the header plus a command. 
* I would expect the header to be the same as in the ATSH command (ATSH8010**F1**)
* The command is a data request 0105 (Engine coolant temperature)
  * I would rather expect a 3E (Tester present) here since that is the intended message to keep the communication going.

So my suggested wakeup message command would be:
> ATWM8010F13E

kavol, 2015-12-16:
so here's the log level debug:
http://uloz.to/xhH6ofoJ/androbd-2015-12-15-zip

- sorry it's a bit messy, I didn't start the engine and it apparently interrupted the communication after a while and also I was switching functions wildly ... hope you'll find what you need

^ this is with "ATWM8010F13E" in the custom config

fr3ts0n, 2015-12-17:
Thank you for the log. Yes I have found what I needed.

Unfortunately your ECU does not support modes 09 (Vehicle info) and mode 0A (Read permanent DFCs)  requests.
It responds with **7F0911** and **7F0A11**, which means **Service (0x09/0x0A) not supported**

* The next release will show a descriptive status message on negative protocol responses.

Other than that, it looks like you are up and running for all your ECU supports :+1: 
If you agree, I'll close this issue.

closed 2015-12-27 by fr3ts0n

kavol, 2016-01-07:
yes, now it talks to my car, thanks a lot

however, there's another issue due to recent changes ... I'll open a new ticket for that