fr3ts0n/AndrOBD #11: Some enhancement ideas

peturdhoye, 2015-10-16:
1) save connected adapter(s) for (2)
2) auto-connect at startup and reconnect when connection lost (eg due to engine off)
3) save selection of data sources for dash/graphs and reselect at app startup
4) better scaling and placement of dash (on my N7 with 4 dials, all 4 stay on 1 row, only half the screen used)
6) larger fonts depending on screen

fr3ts0n, 2015-10-20:
Hi,

Thanks for your ideas.

I will check into these...
Just to keep in mind: 
I would guess that you'd like these memorized presets because you probably always connect to the same adapter- and vehicle. Others (like me) connect to different adapters and vehicles quite frequently. 

Therefore it's easier for me to confirm the adapter from a list, rather than to change the setting of the adapter to be used. Another vehicle also supports different data items, which may not match with the pre-selected ones.

I'll try to find a (hopefully smart) way to have these settings / selections optional.

4 and 6: agreed (do I miss a 5 here?)

peturdhoye, 2015-10-20:
Ah yes, use-cases :)
I'm indeed in the one adapter, one car use-case, I can understand now how you came to your setup.
Maybe something to load and save a setup (adapter+data) and an option to load such a setup on startup of the app.

And there was no 5, or at least there is but it got the number 6. I guess I can only count to 4 :D 

Thanks for your effort, btw!

fr3ts0n, 2015-10-26:
items 4 and 6 implemented with V1.3.2

peturdhoye, 2015-10-26:
\o/

fr3ts0n, 2015-10-31:
items 1, 2, 3 implemented with V1.3.3

closed 2015-10-31 by fr3ts0n