fr3ts0n/AndrOBD #14: protocol error infinite loop

kavol, 2016-01-07:
Trying to use AndrOBD with my car (see issue #10), version 1.3.9 crashes.
It displays an error, see the screenshot, many times and then it closes, presumably exhausting resources.

[image]

kavol, 2016-01-07:
logs here: http://uloz.to/x78n1mJN/androbd-16-01-06-zip
(this tracker still doesn't accept zip despite it says it does ...)

closed 2016-01-08 by fr3ts0n

fr3ts0n, 2016-01-08:
Thanks for the report.

Yep. this was a loop condition, since the service request was not aborted with the NRC.
The loop was triggered with the ELM prompt which wasn't part of my ELM simulation for testing ...

fixed with V1.3.11

commit 31b68e81 (2016-01-08): started V1.3.11
fixed #14 - cancel current OBD service on NRC response (to prevent error loop)
commit 27cb5baa (2016-01-30): started V1.3.12
fixed #14 - initialize ELM adapter on NRC (configurable)