fr3ts0n/AndrOBD #15: no data after protocol error

kavol, 2016-01-14:
Hi.

Thanks for fixing #14 however it still isn't perfect ... After getting such error, which happens either by trying to get vehicle info or by trying to read permanent error codes, the application no longer displays any other data. I have to disconnect and reconnect again before it is able to read sensor values.
Log here:
http://uloz.to/xgN83eKe/androbd-2016-01-14-zip

fr3ts0n, 2016-01-18:
In your log with the attempt to service 0A (permanent fault codes) there is still one data request answered correctly after the NRC was triggered. Also the request 09 got re-sent and answered before the service was stopped. Then, after a pause, all requests respond with NODATA.

It looks like the adapter stops sending the wakeup/tester present messages after a negative response is received. This results in a diagnostic session timeout.

If above assumption is right, a new request should be accepted by the ECU if it is issued before the session times out (after about 2 seconds).
I could imagine a workaround for that by adding an optional reset- and re-initialize on NRC reception.

But since I hate to do changes on guesses: 

Is there a chance you could confirm following case?
* switch to a valid mode (i.e. read data) within 2 seconds after NRC pops up.

Thanks

kavol, 2016-02-05:
not sure if I was quick enough but after first attempt to switch back, it worked, however after second try, it didn't even print the error message about unsupported mode and just stopped getting data

log: http://uloz.to/x9VevyNx/androbd-2016-02-02-zip

fr3ts0n, 2016-02-05:
Yes, it really looks like the ELM "forgets" about the cyclic tester present message after it receives a NRC from the ECU. I have added an OBD configuration item "**Reset on NRC**" which initiates a ELM reset, in case of a NRC response. This reset will be followed by a complete re-initialisation (including custom init commands) which also sets up the cyclic "tester present" message again.

I have tried it first on application level, but it took too long until the reset was triggered, so I moved the handling to protocol level. 

Fix will be available on F-Droid as V1.3.12 soon.

fr3ts0n, 2016-02-05:
Your second attempt did not trigger any NRC any more, because the protocol already has timed out at that point (because of missing, cyclic tester present messages) and so all your further requests have been ignored with NODATA response.

kavol, 2016-02-16:
I've retried with 1.3.12 again today, and it doesn't work well - it disconnected after second attempt to switch to anything else than displaying OBD data

however, I cannot see any new logfiles ... looks like issue #9 has reappeared; note that I had restarted my phone since the upgrade, so any possible stale file descriptors (writing to deleted file etc.) should have been cleaned

fr3ts0n, 2016-02-20:
It's hard to analyze a report without log files.

I checked the logging on my devices again and it logs just fine ...
Is it possible that your "restart" changed the primary storage from internal memory to external sdcard or vice versa?

commit ee2a15a0 (2016-02-05): fixed #15
changed ELM reset on NRC to immediate protocol reaction rather than application reaction since too much delay time
closed 2016-08-21 by fr3ts0n