fr3ts0n/AndrOBD #16: package javax.swing does not exist

mittalharsh54, 2016-01-27:
I got the error package javax.swing doesnot exist after importing the repo in android studio.How to fix them?

mittalharsh54, 2016-01-27:
I have removed the classes you mentuioned in the one issues and it worked. But i want to know whether it will effect the app after removing those files and how.

fr3ts0n, 2016-01-30:
The repo also contains files for several java swing applications for PC (JObd / JaVAG)
The corresponding directoires should be excluded for the AndrOBD build.

You can find the requred excludes in the ant build file **build.xml**

closed 2016-01-30 by fr3ts0n

frankvg, 2016-05-29:
I must be not seeing it, but looking at the build.xml file, I don't see what directories and files to exclude for Android to get rid of the javax.swing errors. Can you list them?  As it seems the package is just for Android, might be nice to remove those files that prevent builds, but just knowing what they are is ok too.  You might even add a comment within those files like "Delete this file to run on Android". Thanks!

fr3ts0n, 2016-05-29:
The files / directories to be excluded are listed in the project.properties file:
android.package.excludes=com/fr3ts0n/common/,com/fr3ts0n/pvs/gui/,com/fr3ts0n/ecu/gui/application/,com/fr3ts0n/ecu/prot/vag/,com/fr3ts0n/prot/gui/

This file is normally used with the ant build system, so the files should be excluded automatically.
Please could you check if your project.properties file got modified / overwritten by some configuration step.

I currently do not intend to remove the files from the repo, since the swing application builds / runs from the same code base.

frankvg, 2016-05-29:
Thanks - just what I needed. I really appreciate the fast response too!  For some reason the exclusions didn't import into Android Studio 2.1.1 Maybe a bug in Studio.