fr3ts0n/AndrOBD #18: Include a Changelog Link, within the F-Droid App', on Info' Page; 1.3.12.

locutis-of-borg-1999, 2016-02-22:
Include a Changelog Link, within the F-Droid App', on Info' Page.  Not present, as of: 1.3.12.

fr3ts0n, 2016-02-22:
Already available. Pressing on the Version number within the APP leads you to the project releases page.

locutis-of-borg-1999, 2016-02-22:
Oh, so it shouldn't be too strenuous, by the next release, to add that link in the F-Droid app's info' page, under "links".

It's good to see what's been fixed.  People uninstall app's, over issues, and look for changelogs, even prior to first install.

I suppose you could tell them, in the description, to "just install it", and look for the hidden link.

locutis-of-borg-1999, 2016-02-22:
https://f-droid.org/manual/html_node/Changelog.html#Changelog

alanthehat, 2016-05-04:
I too would like to see a [changelog] link on F-Droid

fr3ts0n, 2016-05-06:
added metadata change to app profile in f-droid repo and created a merge request
I hope this does the job ...

closed 2016-05-08 by fr3ts0n