fr3ts0n/AndrOBD #19: Cannot detect ECU

gege2b, 2016-02-22:
Hi there
I bought an OBDII dongle ([this one ](http://www.amazon.fr/KKmoon-Bluetooth-interface-adaptateur-compatible/dp/B007XPPKVO/ref=pd_sim_147_6/276-2759162-9462703?ie=UTF8&dpID=41rEry-3rmL&dpSrc=sims&preST=_AC_UL160_SR160%2C160_&refRID=1HEG26AMNK06R9V1GHQE) ) 

So I can look at my car OBDII data (for now, I'm just curious)
My car is a Dacia Logan 1.6 MPI 90 HP from 2006, with the OBDII connector in front of the passenger side
[Infos on OBD support here](http://www.outilsobdfacile.fr/liste-vehicule-compatible-obd2/dacia)

my Android 4.2 phone connects successfully, but then, it freeze a few seconds on "ECU Detect..." and disconnects.

I tried several protocols, differents options but this is always the same (with some variant in the timing)

Engine was always on

[Here is a full log of several trys](https://framabin.org/?fe626b6005a7cc13#Pz4FfvCV2ANE3TX1Q+lN1g9dv6eQidKwb9bXgnOxAVA=)

Tell me if you need more infos

Note : may be the log will show it but the app crashed a few time when in the "developpement option", taping on data customization items, I don't think it's related

cheers

gege

fr3ts0n, 2016-02-22:
Based on the information from the table of your provided link please re-try with following settings (OBD options):

* Secure connection: DISABLED

* Preferred OBD protocol: ISO 14230-4 LWP (5 Baud init)

* ELM adaptive timing: DISABLED

* ELM custom init: add following 3 lines
> ATSH8010F1
> ATWM8010F13E
> ATSW10

* RESET on NRC: ENABLED

That should (hopefully) make it work.

gege2b, 2016-02-23:
Hi there, thank you for the answer

Unfortunatly, still doesn't work

I'm not used to OBD so I can't really help... All I can do is to give you the logs

[Here are they](https://framabin.org/?92c6f52da880431f#JQUmITdZUlcb+BweFdnYNpgG81btN1Ck8qJ9XCbLVK0=)

fr3ts0n, 2016-02-23:
Your adapter is definitely a incomplete ELM327 clone (which is pretty obvious looking at the price), since it doesn't seem to support the full command set. (In the logfile the **ATWM** command, which defines a cyclic wakeup/tester present message, gets rejected with answer '**?**')
Since the custom init commands are only supported partly and are valid for KWP protocol only, it's probably better to delete them again and go with the adapter defaults.

I checked into the blafusel page to verify the protocol for the Logan 1.6 MPI:
(https://www.blafusel.de/obd/obd2_scanned.php?order=usb#2084)

It lists it as ISO 9141 - and also ISO 14230-4, so we should try at least both protocols with all possible init methods, and even again with preferred protocol AUTO.

After deleting the custom init commands try with following preferred protocols:
* AUTO
* ISO 9141-2 (5 baud init)
* ISO 14230-4 KWP (5 baud init)
* ISO 14230-4 KWP (fast init)

PS: have you ever successfully tried the adapter with any other OBD-App (i.e. Torque Light) to verify the adapter actually can do it?

gege2b, 2016-03-04:
Hi, sorry for the late response

I tried Torque Lite without success too, same symptoms

I just tried all you recommendations but without success...
I noticed that all the protocols you listed results in an "OFFLINE" status, except the ISO 14230-4 KWP (5 baud init) wich just disconnects. I don't know if it's important anyway

This time I don't have usable logs, as I forgot to check the "debug" level...

Another thing I noticed while looking at blafusel, is that apparently, my car use the ELM323 interface instead of ELM327, may it be the cause of this issue ?
I will test the dongle on another car as soon as I can

If not, then I think you're right and my dongle is a bad one, I let you close this issue if you think there is nothing more to do on your side

And again, thank you for your help !

closed 2016-08-21 by fr3ts0n