fr3ts0n/AndrOBD #20: Can't load custom data fields from Camry Hybrid

mangolazi, 2016-03-29:
Hi, I'm trying to load hybrid system PIDs from a Camry Hybrid but nothing shows up in the OBD Data screen. Here is my custom data items file:

`svc	pid	ofs	len	bit_ofs	bit_len	bit_mask	formula	format	min	max	mnemonic	label	Remarks
0x21	0x61	0	1	0	8	0x00	ASCII	%s			cust_mg1	MG1 rpm	MG1 rpm
`

It's a bit of a hack since I haven't figured out the correct offsets and data lengths yet. I'm using Prius Gen3 PIDs which are known to work with the Camry from these sites:
[http://priuschat.com/attachments/geniii-prius-4-24-12-zip.37079/](url) Priuschat: Custom PIDs for Torque app
[http://www.scangauge.com/support/x-gauge/toyota-specific/](url) ScanGauge: Prius Gen3 custom fields

fr3ts0n, 2016-04-02:
Hi,
Sorry, but currently AndrOBD only supports OBD related services.
Service 0x21 (RDBLI) and 0x22 (RDBI) are KWP services which also require physical addressing rather than functional addressing which is used for OBD.

I think, in order to support this I would need to extend:
* data model to include physical address
* UI to support additional services

mangolazi, 2016-04-04:
I did a bit of a hack by adding a new 0x21 service and it works, although I needed to set a custom header init string like ATSH7E2. You're right, the main issue is to extend the data model to include non-OBD services and allow custom headers for each query, like how the Torque app does it. 

I haven't gone through all your code but a lot needs to be changed, especially with the logging. I'll see what I can do and I'll send the changes over.

closed 2017-03-17 by fr3ts0n