fr3ts0n/AndrOBD #21: Wifi?

listlfa, 2016-04-15:
Is it possible to add wifi connectivity? Would it be a complex change to the app? I'm willing to learn how to do it if the change can be based off the existing Bluetooth code. Thanks.

fr3ts0n, 2016-04-17:
Yes, adding wifi-based adapters is definitely possible and should not be that big of a deal. 
This requires a new CommHandler implementation which handles a TCP network connection and -stream. 

However, I currently do not have any wifi based adapter available, which makes development, and especially the testing part, quite poor.
I'll see if i can get one for a reasonable price (any suggestions?), or if somebody is willing to donate one of these to get this done ...

listlfa, 2016-04-17:
I've done no Android development before but I'll get your code, get it running, and then buy a wifi adapter. I have a job though, so don't expect any progress at all in the first fortnight at least please :)

fr3ts0n, 2016-04-18:
Implemented initial changes for WIFI support. TCP Connection with elm simulator passed.

fr3ts0n, 2016-04-25:
released with V1.4.0

closed 2016-04-25 by fr3ts0n