fr3ts0n/AndrOBD #22: Convert Intellij to android for AndrOBD

carpiko1010, 2016-08-04:
this andrOBD project is made on intellij, when i import it in android studio, it doesnt recognize the libraries jar files and throws this error.
Error:(21, 19) error: package javax.swing does not exist
Error:(26, 44) error: package javax.swing does not exist
Error:(80, 22) error: cannot find symbol class Icon
Error:(116, 21) error: package javax.swing does not exist
Error:(117, 21) error: package javax.swing does not exist

carpiko1010, 2016-08-05:
I got help from your solution that you provided to exclude the jawa swing calsses.

fr3ts0n, 2016-08-05:
Sorry for the late response. 
Please see answers on issues #1 and #16.
I will add this to the README for future wiithin rhe next days

closed 2016-08-05 by fr3ts0n