fr3ts0n/AndrOBD #23: How to show mileage of car?

carpiko1010, 2016-08-05:


fr3ts0n, 2016-08-05:
The OBD standard does not provide any PID (data item) which contains the absolute mileage / odometer reading of a vehicle. 

The PIDs which come closest to this information are:
* PID 0x21 Distance since MIL activated
* PID 0x31 Distance since ECU reset

carpiko1010, 2016-08-05:
But in other obd android app, they show mileage of car aslo like in torue and obd car doctor.
Also one last query, when we click on obd data it only shows 24 obd items and does not scroll anymore. But as I can see under obd settings> display obd data, there are around 78 of data to be displayed to choose from , but the obd data does on show all of them.

fr3ts0n, 2016-08-05:
Other apps which may display odometer reading most likely use additional diagnostic services to read out the odometer value. However these diagnostic services are vehicle / manufacturer specific, and not part of the standardized OBD diagnostics.

The difference between Configuration screen and data display screen is:
* The configuration screen shows all OBD PIDs (data items) which are **known to the AndrOBD app** and allows you to disable the ones which you are not interested in for the data display.
* The data display shows all OBD PIDs (data items) which are **supported by your vehicle ECU's OBD diagnostic protocol implementation** (if they are enabled within the configuration).

closed 2016-08-21 by fr3ts0n