fr3ts0n/AndrOBD #24: How do I test and execute on the local environment?

chw333, 2016-08-19:
How do I test and execute on the local environment? 

Especially android studio

fr3ts0n, 2016-08-20:
Import into Android Studio. 
Build. 
Execute on device, or Emulator. 

There is a demonstration mode available, if no Elm Adapter is connected.

chw333, 2016-08-21:
Invalid VCS root mappings
The following directories are registered as VCS roots, but they are not: C:\Users\woong\Documents\AndrOBD\AndrOBD.wiki C:\Users\woong\Documents\usb-serial-for-android

this kinds of error is continuously happen.

chw333, 2016-08-21:
I'd like to keep in touch with you fr3ts0n
pooh0216@gmail.com

chw333, 2016-08-21:
I'd like to do some codereview with you and improve the application.

Thanks.

fr3ts0n, 2016-08-21:
Thanks for your finding on the VCS mappings...
I did not notice since I do have these links mapped to allow to have the wiki available within the same IDE project.

You are welcome to review and comment for any improvements.
I would suggest you to comment / change the issues you have found so far and create a pull request. 
Then we can discuss and implement these changes.

Thx

chw333, 2016-08-21:
" I did not notice since I do have these links mapped to allow to have the wiki available within the same IDE project. " you mean I have to link with wiki?

please more specifically explain. and I'd like to keep in touch with you. because improve this app.

Thanks and Best wishes

chw333, 2016-08-21:
Not to translate this project. only focused on function improvement

fr3ts0n, 2016-08-21:
No, I'll remove the references from the Project's VCS mappings, since it is not necessary.

chw333, 2016-08-21:
yeap :+1:

chw333, 2016-08-21:
Is it possible to communicate with you using other things?
Don't be narrow mind. my major is computer science. and I'd like to cooperate with you. be open mind.

chw333, 2016-08-21:
Got it thanks

closed 2017-02-15 by fr3ts0n