fr3ts0n/AndrOBD #26: Cannot export into CSV

BobSquarePants, 2016-10-23:
Hi people,

The exported file format is .obd and I don't know with what open them ?

If have found in settings\Development option 2 related option 
[image]

but when i click on it I have
[image]

Any idea ?

fr3ts0n, 2016-10-24:
The settings dialog above is used for customisation/expansion of protocol parameters. 
(see: https://github.com/fr3ts0n/AndrOBD/wiki/Data-customization)

The .obd file can be loaded with AndrOBD menu function "Load measurement"
After loading the .obd you can select the parameters you want to see and chart these out ...
CSV format of selected data can be shared from the chart view (using share button)

BobSquarePants, 2016-11-01:
Thank you @fr3ts0n !

So I can send myself an email with the .CSV, but strangely the csv is not an attached file, instead all the content of the csv is write in to the body of the email without any CR.
[csv.gif](https://postimg.org/image/60w2vn5hf/)

And also is there anyway the change the first Columns (second from start) as a full time like yyyy-mm-dd HH:MM ?

Thx and good job btw !

fr3ts0n, 2016-11-06:
Thanks for your input.

Yes, you are right, it does make more sense to export it directly to a file which then may be shared, or whatever ...
CR is contained within the current output format, but it is not a CR/LF as windows programs would expect.
Also the full time output makes sense.

I will try to find a configurable solution to allow adaptions to the CSV output. Please give me some time to think about this ...

fr3ts0n, 2017-01-15:
fixed with release V1.4.6

* field and line delimiters configurable
* export into CSV file
* immediate CSV file share configurable
  - shared as file attachment

closed 2017-01-15 by fr3ts0n