fr3ts0n/AndrOBD #28: Unable to get data directly after being connected to the car

maximusTheSituation, 2017-04-13:
This project is very useful, I am able to deploy it locally on Android Studio but I'm getting stock when I want to get data directly from the MainActivity without going on another views like the diagnosis one. Let me explain, Basically, I want to access the data directly right after the bluetooth connection. The goal is to send them in a webserver on a periodic fashion. That means as soon as the user is connect to his car, I pull data continuously and send them to the webserver. So I can define a class for these data, let's call it `OBDReading`. And a simple structure of this class can be:

```
public class  OBDReading {
	private String OBDLabel;
	private String OBDValue;

	public String getOBDLabel() {
		return OBDLabel;
	}

	public void setOBDLabel(String OBDLabel) {
		this.OBDLabel = OBDLabel;
	}

	public String getOBDValue() {
		return OBDValue;
	}

	public void setOBDValue(String OBDValue) {
		this.OBDValue = OBDValue;
	}
}
```
Can anyone help me to achieve this?

fr3ts0n, 2017-04-16:
For automatic connection / data display you may want to use the Feature in Settings: "Use last settings" to automatically select:
* last ELM adapter
* last vehicle connection
* last data selection

For Issuing / forwarding data changes to some server, you may want to attach a PvChangeListener to the data items in ObdProt. Your ChangeListener will be notified on all changes of OBD data items which it can then forward to a server.

fr3ts0n, 2017-04-16:
[image]

maximusTheSituation, 2017-05-06:
Ok thank you for the response. I solved the isssue

closed 2017-05-06 by maximusTheSituation