fr3ts0n/AndrOBD #30: Enable Hebrew translation in Weblate

yarons, 2017-09-21:
Awesome project, can I pitch in?

Thanks!

fr3ts0n, 2017-09-21:
You can just start a new translation for hebrew language on weblate.org.
You just have to register as a weblate user for this.
There is nothing I have to enable for that.

Thank you for joining the team :)

yarons, 2017-09-21:
Are you completely sure about that? Because I'm already registered and I couldn't add Hebrew to the list, should I try and contact Michal?

fr3ts0n, 2017-09-21:
I have just added the translations for hebrew to the weblate project.
* Strings
* Fault code list
* Data descriptions

I'm not sure why you havent been able to do this, since other people did this for other languages before.
I just logged in and added a new translation at the bottom of the language list.

However, now the hebrew translation is available and ready for you to start :)

closed 2017-09-24 by fr3ts0n