fr3ts0n/AndrOBD #31: Detecting ECU… is misleading

yarons, 2017-09-21:
I'm currently using the following plug:
https://bafxpro.com/products/obdreader

When I first connected to it I got a message saying the the ECU (Not sure what that is) is in detection process, I just waited, but then I tried playing with the screen and I realized I can already use the app and make dashboards etc. so this message was not a blocker and I'm guessing that at this point my ECU was correctly detected.

Is there any way to provide more data about that? It happens each and every time.

fr3ts0n, 2017-09-21:
Thanks for your report.

Please could you provide some debug log files for further analysis of the vehicle communication.

Please see wiki page for details on setting logging depth and log file location:
https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues

Thx

fr3ts0n, 2018-01-03:
This was caused by the possible delay of Android Toast messages. 
The message appeared several seconds after it has been issued by the event.

Fixed by changing most status/event notifications to be displayed within the toolbar status. 
Toolbar will be updated (and made visible if hidden) immediately when there is any status change.

-> **AndrOBD V1.5.5**

closed 2018-01-03 by fr3ts0n

yarons, 2018-01-03:
Thanks a lot!