fr3ts0n/AndrOBD #32: Problem with backwards fault codes...

glockpod, 2017-10-27:
I am running AndrOBD V1.5.2 on a Nexus 7 tablet (pre 2012 model) and a purple ELM 327 compatible bluetooth adapter.  My subaru threw 4 fault codes according to an Equis stand alone OBD reader and 3 other android OBD app readers.  All of the android apps were using the same purple ELM adapter.

All readers said the codes were P0301, P0302, P0303, P0304.  Which are misfire cylinder 1-4 (which matches the car's symptoms).

AndrOBD V1.5.2 found 4 fault codes too, P0103, P0203, P0303, and P0403... Which, except for the P0303 code, are nonsense.

It would appear that somewhere the fault code number is getting reversed.

-Chuck

fr3ts0n, 2017-10-27:
Thank you for reporting this.
Yes it really sounds like there is a reversion of the numbers ...

Please could you provide a debug log file to verify the exact cause of this problem?
see: https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues
for adjusting the logging information details to FINE.

Thank you.

fr3ts0n, 2017-10-30:
I found out that since I changed to the new logging method the log files only get written, if the logging path already exists (which it did on all my test devices).

just added new issue #34. 
I'll fix that immediately.

After that you should be able to find the log files again so we can analyze your problem ...

fr3ts0n, 2017-11-24:
Do you still have a chance to create a log file output of this issue. With the most current version it is possible again ;) 

I suspect a misinterpretation of the fault code output, since this has an optional parameter of number of reported fault codes. If the app falsely interprets the presence of this parameter, then the following fault codes would be interpreted with a shifted byte offset of 1 which looks very similar to a byte swap.

Thx

closed 2018-11-03 by fr3ts0n