fr3ts0n/AndrOBD #33: AndrOBD app stopped when connecting to ELM327

cnieves1, 2017-10-28:
Hi,
I tried an ELM327 bluetooth adapter on a Seat Leon 1.6 2001 model. I think this car has a K-line bus. 
AndrOBD app is stopped after the Bluetooth adapter was selected, when trying to connect.
I could not find any log, even after I tried again with log level set to All.
I also tried setting the interface to K-line instead of automatic detection, but same result.
What else could I do to know what is wrong?

Thanks in advance

cnieves1, 2017-10-28:
Just in case, I used Android 6.0.1.
I also tried Torque Lite with the same ELM327 adapter and everything seems to work fine.

fr3ts0n, 2017-10-28:
Thank you fro reporting the issue.

It might be worth a try to switch off **Adaptive ELM timing** option in OBD settings.
This might be the cause of communication timing out too early on slow communication protocols.

Other than that, I really would need a log file for more detailed analysis.

For log file location and activating debug logging, here is the corresponding wiki page:
https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues

cnieves1, 2017-10-29:
I'm afraid I already tried that... 

Yes, I was aware of the wiki page about log files. There are folders with similar names in /Android/data, but there is no folder named like that.
I also tried searching *obd* and *OBD* with no success. 

I also checked that this app has storage permission.
Android shows this app is using 60KB od data and 16KB of cache. But... How can I find where are those files?
Only delete data and delete cache options are shown...

Could be something related with log file creation? What happens in the app before log file is created?

Thanks!

fr3ts0n, 2017-10-30:
I found out that since I changed to the new logging method the log files only get written, if the logging path already exists (which it did on all my test devices).

just added new issue #34. 
I'll fix that immediately.

After that you should be able to find the log files again so we can analyze your problem ...

cnieves1, 2017-11-07:
Hi, 
I had to wait until F-droid package was available.
Just for your interest, the log in my system is not in /Android/data as other app's data, but in / directly.
Regarding the crashing, I tried several times. Sometimes the app told me that it can not connect. Sometimes the app was stopped.
Here are some logs, but I can not see something wrong there...

[AndrOBD.log.2.txt](https://github.com/fr3ts0n/AndrOBD/files/1449011/AndrOBD.log.2.txt)
[AndrOBD.log.3.txt](https://github.com/fr3ts0n/AndrOBD/files/1449012/AndrOBD.log.3.txt)
[AndrOBD.log.4.txt](https://github.com/fr3ts0n/AndrOBD/files/1449013/AndrOBD.log.4.txt)
[AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/1449014/AndrOBD.log.0.txt)
[AndrOBD.log.1.txt](https://github.com/fr3ts0n/AndrOBD/files/1449015/AndrOBD.log.1.txt)

Thanks,

fr3ts0n, 2017-11-07:
I can not find anything obvious in the log files either.

Did you check for any system error messages (logcat) during connection phase of AndrOBD?

I can imagine 3 things to try:

1. Switch on Bluetooth manually, before starting AndrOBD to bypass timing issues during system BT initialisation.

2. It may be a chance to try using secure bluetooth communication rather than the default:
> OBD settings -> Secure connection -> ON

3. Another chance may be to turn OFF adaptive ELM timing (as suggested above)
> OBD settings -> Adaptive ELM timing -> OFF

cnieves1, 2017-11-09:
Hi,
I tried switching on Bluetooth before starting AndrOBD, without success.
The same with secure communication.

I thought I tried 3, but reviewing the attached logs, it seems not. I will try again.
[AndrOBD.log.1.txt](https://github.com/fr3ts0n/AndrOBD/files/1456619/AndrOBD.log.1.txt)
[AndrOBD.log.2.txt](https://github.com/fr3ts0n/AndrOBD/files/1456620/AndrOBD.log.2.txt)
[ELM327_Identifier_log0.txt](https://github.com/fr3ts0n/AndrOBD/files/1456621/ELM327_Identifier_log0.txt)
[AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/1456622/AndrOBD.log.0.txt)

I suspect AndrOBD is expecting the adapter to answer some AT command, and it gets no answer...
I also tried ELM327 identiifer, to see what commands are supported by my adapter (ELM327 Mini). Attached is the log. Some of the AT commands may depend on the vehicle it was attached, so remember it was a ISO9141 only.
The logs always end with RX thread started. Could you please tell me what AndrOBD is doing, and what AT commands is using and what to expect?
I can use a Bluetooth terminal in order to emulate AndrOBD AT commands, and see what happens...

fr3ts0n, 2017-11-09:
What I am missing in your logs is the attempt of sending anything out to the adapter (like the initial ATZ command) after the bluetooth is connected.

Here is a communication log of a typical initialisation sequence for reading data: 
```
2017-11-09      06:25:41.243    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'ATZ'
2017-11-09      06:25:41.286    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'ELM327v2.1'
2017-11-09      06:25:41.294    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:41.295    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'ATE0'
2017-11-09      06:25:41.326    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'ATE0'
2017-11-09      06:25:41.329    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'OK'
2017-11-09      06:25:41.332    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:41.332    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'ATL0'
2017-11-09      06:25:41.358    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'OK'
2017-11-09      06:25:41.360    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:41.361    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'ATS0'
2017-11-09      06:25:41.387    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'OK'
2017-11-09      06:25:41.389    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:41.389    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'ATSP0'
2017-11-09      06:25:41.417    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'OK'
2017-11-09      06:25:41.420    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:41.422    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'ATH1'
2017-11-09      06:25:41.450    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'OK'
2017-11-09      06:25:41.453    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:41.454    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'0100'
2017-11-09      06:25:43.289    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'SEARCHING...'
2017-11-09      06:25:43.930    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'7E8064100BE3EB81B'
2017-11-09      06:25:43.934    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:43.934    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'ATH0'
2017-11-09      06:25:43.965    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'OK'
2017-11-09      06:25:43.968    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:43.977    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'0100'
2017-11-09      06:25:44.070    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'4100BE3EB81B'
2017-11-09      06:25:44.075    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:44.075    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'0120'
2017-11-09      06:25:44.162    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'41208005A011'
2017-11-09      06:25:44.166    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:44.167    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'0140'
2017-11-09      06:25:44.262    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'41407C020400'
2017-11-09      06:25:44.343    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:44.347    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'0101'
2017-11-09      06:25:44.440    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'410100076100'
2017-11-09      06:25:44.540    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'>'
2017-11-09      06:25:44.541    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 TX:'ATST31'
2017-11-09      06:25:44.573    FINE    com.fr3ts0n.prot        com.fr3ts0n.ecu.prot.obd.ElmProt@3536fcc3 RX:'OK'
```

fr3ts0n, 2017-11-10:
I have a suspicion on the missing ATZ command...

I found a small time window during AndrOBD initialisation which may prevent the ELM initialisation to be initiated after a bluetooth connect.

I have created a test version for you to check/verify if the problem disappears with this change/fix in the init sequence.

https://drive.google.com/file/d/18T-6K7xv0D1xSnvEiujbKZbcdTEQBCkB/view?usp=sharing

Please note that you will have to uninstall the F-Droid version first, since this test version just fell out of my local build w/o a F-Droid signature.

Thx

cnieves1, 2017-11-10:
Tried. Attached log. It doesn't work yet but there is another line in log after rx thread started. 
I guess you are right and there is something wrong with elm327 initialization... 
Can I reproduce the commands in a bluetooth terminal and send you the output? What has I to type?

Thanks

2017-11-10	19:11:06.678	INFO	AndrOBD	AndrOBD V1.5.4 starting
2017-11-10	19:11:07.180	INFO	com.fr3ts0n.prot	Reset on NRC = false
2017-11-10	19:11:07.188	INFO	com.fr3ts0n.prot	ELM min timeout: 12 -> 12
2017-11-10	19:11:07.189	INFO	AndrOBD	Conversion: metrico
2017-11-10	19:11:07.190	INFO	com.fr3ts0n.prot	Preferred protocol: Automatic
2017-11-10	19:11:07.366	INFO	AndrOBD	Conversion: metrico
2017-11-10	19:11:27.815	INFO	com.fr3ts0n.prot	Preferred protocol: ISO 9141-2 (5 Baud Init)
2017-11-10	19:11:33.949	FINE	CommService	connect to: 00:1D:A5:68:98:8D
2017-11-10	19:11:33.962	FINE	CommService	setState() NONE -> CONNECTING
2017-11-10	19:11:33.968	INFO	CommService	BEGIN mBtConnectThread SocketType:Secure
2017-11-10	19:11:36.638	FINE	CommService	connected, Socket Type:Secure
2017-11-10	19:11:36.639	FINE	CommService	create BtWorkerThread: Secure
2017-11-10	19:11:36.648	INFO	CommService	BEGIN mBtWorkerThread
2017-11-10	19:11:36.650	INFO	stream	RX Thread started
2017-11-10	19:11:36.660	FINE	CommService	setState() CONNECTING -> CONNECTED

fr3ts0n, 2017-11-12:
Sorry, but I am just about out of ideas ...

It almost looks like the AndrOBD main message handler does not process the messages from the connection thread as it should, but I don't have any idea why it doesn't.

Does the action/status bar update the status during- and show status "CONNECTED" after this connection attempt. (This should be visible as sub-title in the action bar)

Is it possible to select "OBD data" mode in this state?
What happens if you select it?

Other than that, I have a lot of questions about your device:
Could you please describe details of your environment. (device type, android type, version, custom build version?)

Do you have the chance to check for system error messages during AndrOBD startup via "logcat"?

Thx

cnieves1, 2017-11-14:
Attached is the output of 'logcat -f log.txt'.
This is a MotoG3 phone, with Motorola's firmware, and Android version 6.0.1. It is not a custom build version.

AndrOBD is stopped and closed. 
[log.txt](https://github.com/fr3ts0n/AndrOBD/files/1469853/log.txt)
I didn't see any update to the status bar, but I can't promise. I can try to make a video if you want...

Regards,

cnieves1, 2017-11-14:
Ups. What I did was run 'logcat -f log.txt' in a terminal on the phone, then run AndrOBD. It crashed after a while, so I run it again, and it crashed again, but sooner.

If I connect to ELM327 with a Bluetooth terminal and type ATZ, I got no answer. However if I run first "ELM327 identifier" app, then do the same with the Bluetooth terminal, it answers. Unfortunately, I got AndrOBD stopped again when running AndrOBD again after ELM327 identifier app.

I guess it needs some initialization. Could be possible that this initialization is wrong, and AndrOBD can't get an answer?

cnieves1, 2017-11-14:
Tried again. Status bar is not updated, and AndrOBD is stopped and closed.
I got something strange with a bluetooth terminal. Here is what happened:

[16:32:16.531] << ATZ
  Here I run ELM327 identifier
[16:33:48.955] << ATZ
[16:33:50.097] << ATZ
[16:33:50.703] << ATZ
  No answer to the above commands
[16:33:56.575] << ATI
[16:33:56.597] >> ATZELM327 v2.1>ATZELM327 v2.1>ATZELM327 v2.1>ATIELM327 v2.1>
  Got all answers together
[16:34:12.394] << ATI
[16:34:12.414] >> ATIELM327 v2.1>
[16:34:15.669] << ATI
[16:34:15.691] >> ATIELM327 v2.1>
  ATI seems to work right.
[16:34:25.054] << ATZ
[16:34:26.789] << ATZ
[16:34:27.604] << ATZ
[16:34:28.341] << ATZ
[16:34:28.972] << ATZ
[16:34:29.581] << ATZ
[16:34:34.636] << ATI
[16:34:34.665] >> ATZELM327 v2.1>ATZELM327 v2.1>ATZELM327 v2.1>ATZELM327 v2.1>ATZELM327 v2.1>ATZELM327 v2.1>ATIELM327 v2.1>
  Again the same. I tried with more time between commands:
[16:35:04.284] << ATZ
[16:35:12.590] << ATZ
[16:35:19.964] << ATZ
[16:35:29.141] << ATI
[16:35:29.170] >> ATZELM327 v2.1>ATZELM327 v2.1>ATZELM327 v2.1>ATIELM327 v2.1>
[16:35:41.041] << ATI
[16:35:41.070] >> ATIELM327 v2.1>
[16:35:44.496] << ATI
[16:35:44.524] >> ATIELM327 v2.1>

Seems a strange behaviour to me... If you tell me what is the initialization procedure, commands and expected answers, I can try with the Bluetooth terminal...

cnieves1, 2017-11-15:
Sorry, I missed the initialization sequence above. Here it is what I found running through Bluetooth terminal.
Plugged the odb2 adapter, then connected to it:[16:31:31.365] << ATZ
[16:31:37.371] << ATE0
[16:32:00.333] << ATEL0
[16:32:06.388] << ATES0
[16:32:26.574] << ATESP0
[16:32:35.692] << ATH1
[16:32:51.397] << 0100
[16:33:13.984] << ATI
[16:33:20.998] << ATI

No answer to these commands. Then disconnected and run ELM327 Identifier app. When it finished, I connected again and:
[16:35:08.619] << ATI
[16:35:08.691] >> ATIELM327 v2.1>
[16:35:13.588] << ATZ
[16:35:18.702] << ATE0
[16:35:18.707] >> ATZELM327 v2.1>ATE0OK>
[16:35:32.311] << ATEL0
[16:35:32.350] >> ?
[16:35:43.788] << ATES0
[16:35:43.835] >> >?
[16:35:56.446] << ATESP0
[16:35:56.451] >> >?>
[16:36:07.732] << ATH1
[16:36:07.736] >> OK
[16:36:16.563] << 0100
[16:36:17.400] >> >SEARCHING...UNABLE TO CONNECT>
[16:36:39.532] << 0120
[16:36:40.376] >> SEARCHING...UNABLE TO CONNECT>

So it seems that this ELM327 Identifier app does something that this adapter recognizes and starts to answer the commands.
 I don't understand why an answer to a command is sent after received the next command, but it seems to work somehow...

Does this help you to understand what is happening? Anything you want me to try?

Thanks,

fr3ts0n, 2017-11-17:
It appears your adapter does something strange on the ATZ command which affects the bluetooth stream to the mobile.

What happens if you start the sequence with ATI (instead of ATZ) after plugging in the adapter without running the ELM327 identifier app before.
- If this gets answered, I will try to create a version which allows to disable the ATZ and replace it with ATI.
- If it doesn't get answered, this means the identifier app does something special on the bluetooth connection which I don't know of. 
Unfortunately it is not an open source app, so I cannot find out what it is ...

cnieves1, 2017-11-21:
I'm afraid ATI is not answered...
This app seems to work for me, at least the initialization. I didn't check further:
https://github.com/twaddington/android-obd-reader

Could you check what are the differences?

fr3ts0n, 2017-11-22:
Thanks for the link.

I'll check out the differences

fr3ts0n, 2017-11-24:
There are not many differences in the way the bluetooth is set up.
It has an additional connection fallback method, but based on the AndrOBD log establishing the connection does not seem to be the problem.
Just in case: if you use the android-obd-reader on your phone, do you get a log entry?
``
"There was an error while establishing Bluetooth connection. Falling back.."
`` 

android-obd-reader also uses the ATZ command to initialize the adapter so this is not the difference.

Further the android-obd-reader works single threaded by polling/waiting for the answer, processing and then sending next request, while AndrOBD sets up separate threads for receiving, processing, displaying and sending. 
That looks like the major difference, since the bluetooth connection may be read and written simultaneously. That may be a problem with your phone's bluetooth impementation, or with the adapter's bluetooth implementation.

Unfortunately I don't see a realistic chance to change that behaviour of AndrOBD, since this would be a rather drastic change to the overall setup which is intentionally designed the multi threaded way.

fr3ts0n, 2017-11-25:
Yes, you are right. 
We should not give up as long as you are willing to support me with your tests.
I have to thank you for all your efforts for this already! Thank you very much.

The frustrating part for me is that I haven't found any indicators where the program flow starts this de-tour and I can only guess it's possible causes, since I cannot re-produce this behaviour at all.
Since it works on thousands of other device/adapter combinations it must be a small detail.

So the challenge for us is to find the needle in the haystack by me directing you where to search and report what you have found ...
Can we find a communication media / setup where we can share the findings and thoughts more efficiently to be able to look over the others shoulder?

My current suspects/thoughts are:
- App closing w/o exception notification
  - Uncaught exception?
    - Should be reported by Android system?
  - Exception caught, but ignored?
    - Should leave the App running?
- Internal signalisation of a "Bluetooth connected" event does not trigger the sending of the adapter initialisation command (ATZ).
  - Why?
    - App closing at this point w/o error message and w/o log entry??
    - App still running, but did not receive the notification??
- Bluetooth implementation of mobile (or ELM adapter) might not handle full duplex messaging (TX while RX)
  - Not very likely, because not even the intention to send the initial ATZ is indicated in log.
    - This would be logged before it would be sent.

cnieves1, 2017-11-25:
I agree. I will try to do my best. 

I'm not an Android developer, so I have not a clue about where to start. Let's start with something and see what we find. 

What comms setup do you suggest? Can you start a private thread so we can share our data?

Is it possible to forward the SPP port so you can access my adapter remotely?

fr3ts0n, 2017-11-25:
Do you use irc, telegram, jabber, or something like this (that also supports a linux client)?

cnieves1, 2017-11-25:
Telegram or IRC is fine for me.
However, this thread is public for aharing details...

fr3ts0n, 2017-11-25:
I have created a channel on telegram for AndrOBD development.
https://t.me/AndrOBD_dev

This should make live easier for sharing details with a shorter turnaround time

cnieves1, 2017-11-25:
Joined

cnieves1, 2017-11-25:
Sorry, I have no permission to write on the Telegram channel..

fr3ts0n, 2017-11-27:
Issue was caused by incorrect format string (s% intead of %s) within spanish translation

closed 2017-11-27 by fr3ts0n