fr3ts0n/AndrOBD #35: Copyright violation

Poussinou, 2017-11-01:
Hi @fr3ts0n 

I found [this app](https://play.google.com/store/apps/details?id=com.ODB2.Scanner.Diagnostic.odbScannerDiagnostic) on the Google Play Store. This guy just took your source code and put it on the Store, without warning this is free software and GNU GPL v3 licensed...

As you are the owner of the source code, feel free to open a request to google [here](https://support.google.com/legal/contact/lr_dmca?product=googleplay) to remove the false app from the Store, if it's your wish ;)

Don't forget to tell Google that:

* The app is licensed under GNU GPL v3 license and this guy is violating it;
* Provide screenshots of your app and screenshots of the fake app;
* He is falsely claiming he made the app.

fr3ts0n, 2017-11-07:
Thanks for your info
I haven't seen this yet.

Even the class structure is mine (without any refactoring), since generated log files show original class names. It  is based on a rather old version (V1.3.x).
Artwork is changed, interstitial ads added and the app is renamed, that's it.
Even some of the Playstore screen shots are fake, since the graph display is not from this app.

DMCA complaint is filed ...

Poussinou, 2017-11-07:
So many open-source/free/libre source code theft on the the Play Store, you cannot even imagine... Have a look at the different Issues I opened in the past 3 months, you will be surprised... (I have dozens of names of fake developers like "Super App For Free" in a log. I kill them one after the other when I can, your DMCA complaint really help me... 👍 )

fr3ts0n, 2017-11-11:
app is removed :+1:

closed 2017-11-11 by fr3ts0n