fr3ts0n/AndrOBD #36: NumberFormatException

tianxiaogu, 2017-11-04:
It seems that `parseInt` is not properly used.

```
[APE] // CRASH: com.fr3ts0n.ecu.gui.androbd (pid 6501) (elapsed nanos: 8183794177457)
[APE] // Short Msg: java.lang.NumberFormatException
[APE] // Long Msg: java.lang.NumberFormatException: Invalid int: "627768533312155715515"
[APE] // Build Label: Android/sdk_google_phone_x86/generic_x86:6.0/MASTER/4088240:userdebug/test-keys
[APE] // Build Changelist: 4088240
[APE] // Build Time: 1497047463000
[APE] // java.lang.NumberFormatException: Invalid int: "627768533312155715515"
[APE] // 	at java.lang.Integer.invalidInt(Integer.java:138)
[APE] // 	at java.lang.Integer.parse(Integer.java:413)
[APE] // 	at java.lang.Integer.parseInt(Integer.java:367)
[APE] // 	at java.lang.Integer.parseInt(Integer.java:334)
[APE] // 	at java.lang.Integer.valueOf(Integer.java:525)
[APE] // 	at com.fr3ts0n.ecu.gui.androbd.MainActivity.setAutoHider(MainActivity.java:794)
[APE] // 	at com.fr3ts0n.ecu.gui.androbd.MainActivity.onSharedPreferenceChanged(MainActivity.java:876)
[APE] // 	at android.app.SharedPreferencesImpl$EditorImpl.notifyListeners(SharedPreferencesImpl.java:479)
[APE] // 	at android.app.SharedPreferencesImpl$EditorImpl.apply(SharedPreferencesImpl.java:387)
[APE] // 	at android.preference.Preference.tryCommit(Preference.java:1415)
[APE] // 	at android.preference.Preference.persistString(Preference.java:1448)
[APE] // 	at android.preference.EditTextPreference.setText(EditTextPreference.java:92)
[APE] // 	at android.preference.EditTextPreference.onDialogClosed(EditTextPreference.java:146)
[APE] // 	at android.preference.DialogPreference.onDismiss(DialogPreference.java:395)
[APE] // 	at android.app.Dialog$ListenersHandler.handleMessage(Dialog.java:1323)
[APE] // 	at android.os.Handler.dispatchMessage(Handler.java:102)
[APE] // 	at android.os.Looper.loop(Looper.java:148)
[APE] // 	at android.app.ActivityThread.main(ActivityThread.java:5417)
[APE] // 	at java.lang.reflect.Method.invoke(Native Method)
[APE] // 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
[APE] // 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
[APE] // 

```

fr3ts0n, 2017-11-05:
What did you do to produce this exception?
* What I read from the call stack, you entered a hughe number as AutoHide timeout ...
What is the further reaction of the app (on further starts)?

tianxiaogu, 2017-11-05:
Thanks for your response.

This exception can be reproduced easily following the instructions below.

1. `Settings -> Common settings`.
2. Enable `Hide toolbar`.
3. Click `Hide toolbar dealy`.
2. Input a very large number.

The app crashed and cannot be restarted unless you clear all data.
It seems that the invalid number string has been saved into persistent data.

fr3ts0n, 2017-11-05:
OK, I'll have to fix that by defining / checking the value to be within a defined range ...

commit 66cf84f4 (2017-11-07): Fix #36: Catch/Log exception of bad preference entries
closed 2017-11-07 by fr3ts0n