fr3ts0n/AndrOBD #38: Increasing frequency of data aquisition

hooger, 2017-11-08:
Hello,

is it possible to make a new setting to increase the frequency of signal reads? As I see it is 5 seconds now which is really sparse.

Thanks in advance and keep up the great work!
hooger

fr3ts0n, 2017-11-08:
Hello @hooger ,

The overall data update cycle is dependent on the number of data items to be read. Per default AndrOBD requests all data items supported by your vehicle cyclically.
This list of requested items can be controlled/limited by selecting (a set of) individual data items and display in a filtered list-, gauge-, head up-, dashboard- or graph display. 

AndrOBD attempts to read the signals at the highest possible rate considering the communication delay factors below. It does not use any additional delays or fixed update rates. 

These communication delay factors are:
- ELM adapter response time / timeout
  - There is quite a big variance in response time between different ELM adapters / clones.
- Communication supported by vehicle 
  - Method (CAN / K-Line)
  - Protocol (ISO / KWP / J1939)
  - Baud rate (10.4 - 500 kBaud)
- Protocol response time of the vehicle

For the ELM adapter timeout AndrOBD provides a feature "Adaptive timing" which adapts the ELM read timeout to the smallest possible value during communication. So communication starts rather slow and speeds up as long as the car is reporting valid data. This feature can be en-/disabled in the OBD settings.

If you need any further support/analysis about your reported timings, please could you provide a log file with level DEBUG / FINER as described [here](https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues).

Thx

fr3ts0n, 2017-11-08:
Sorry, I did not want to close it yet ...

hooger, 2017-11-09:
Thank you for the quick and exhausting answer. I think the high number of requested items is behind the low frequency than.
Is it possible to have a short wikipedia entry about the possible impacts of the listed delay factors?
Thanks in advance!
hooger

closed 2017-11-09 by hooger