fr3ts0n/AndrOBD #40: Trying to add this code in android studio but unable to do that.

mrizwan2018, 2017-11-26:
Hi, i am trying to add this module in android studio but unable to do that.
Please help me in this regard thanks.

fr3ts0n, 2017-11-26:
It is set up as a Android studio project rather than a module.
I use AS 2.3 since AS 3.x again has problems handling non gradle projects.

If you use AS 2.3 just open it as an AS project, since the .idea folder content of the AS project is contained in the repo.

mrizwan2018, 2017-11-26:
Can you please explain how can i setup this code in android studio.
Thanks in advance.

fr3ts0n, 2017-11-28:
It is set up as a Android Studio (AS) project rather than a module.
Since AS 3.x again has problems handling non gradle projects I use AS 2.3.

## Setup sequence
- Get AS 2.3
- Get Android SDK
- Setup Android SDK within AS 2.3
- Clone repository
- Open AS 2.3
- Open Project AndrOBD
- Build APK

softgearko, 2017-11-29:
FYI,  In my case,
I used Android SDK 23.3 (Android 6.0) for SDK Platforms, Android SDK Tools 26.1.1 at SDK Tools in Tools in Android Studio 2.3.3.
At File/Project Structure.../ Platform Settings/SDKs, Build Target: Android API 23. 
It works.

alaaodeh, 2017-11-29:
Hi,

Can you send me screen shots please?
And video if possible ?

engalaaodeh, 2017-12-02:
Hi,

I am still facinv gradl project error ,can any ond please to provide me the project updated and working on any android stuido ASAP?

Regards

fr3ts0n, 2017-12-09:
Please check/verify versions of your Android SDK modules (in SDK manager):
* Android SDK tools: 25.2.5
* Android SDK build tools: 25.0.3

Updates to any newer version of above SDK tool packets will generate build failures, since it is a non gradle project, and newer tool versions fail to build correctly.

Vince-King, 2017-12-16:
Hi, I'm having an install issue.

I have installed AS and have imported the project. When I try an build the project I get errors regarding no javax.swing.... How do you import swing to Android?

I also get build errors regarding freechart not being available...How do you use the java libs in Android?

Thanks

Vince

fr3ts0n, 2017-12-17:
You may want to **OPEN** the AS project rather than to ~~IMPORT~~  it.

The difference is that the existing AS project excludes several sub-directories which are only used to build JDK-based PC application. These existing AS project excludes get overwritten when you ~~IMPORT~~ it.

Vince-King, 2017-12-17:
I deleted a few of my sdk's, removed some tof the additional build tools, and thought I was making progress. Unfortunately, I did not get it working.

I have just uninstalled AS and am about to start again. Ill try OPEN an existing AS project (I thought I did last time).

Thanks

Vince

Vince-King, 2017-12-17:
This time I downloaded 2.3.2 from Androids archives https://developer.android.com/studio/archive.html 
I installed the supplied SDK, and it appears to build :)
[image]

With other Android projects I have worked I was then able to install the app to my device (or emulator), but am now having trouble finding that option. How do I now get to launch the app?

Thanks again

Vince

Vince-King, 2017-12-17:
I also just noticed these messages on launch of AS
13:05	Migrate Project to Gradle?
			This project does not use the Gradle build system. We recommend that you migrate to using the Gradle build system.
			More Information about migrating to Gradle
			Don't show this message again.
13:05	Platform and Plugin Updates: Android Studio is ready to update.
13:05	Update Property Files
			The structure of following Android modules was changed:
			AndrOBD
			Would you like to update related project.properties files?
			Only once
			Always for these modules
			Never for these modules
13:05	* daemon not running; starting now at tcp:5037
13:05	* daemon started successfully

Do I need to migrate to gradle? Can you tell me how to migrate the project to gradle?
Do I need to be updating related project.properties? if so... to what?

Thanks again

Vince

fr3ts0n, 2018-02-24:
With V2.0.0 release AndrOBD repo now is changed to gradle build system.

It now can be imported as gradle project with latest version of Android Studio (currently AS V3.1) using latest Android SDK.

It splits into 3 gradle modules:
* **androbd** - The AndrOBD app
* **plugin** - Android library for common extension plugin handling
* **library** - Java library for common functionality
  - Process variable handling
  - Communication protocol handling
    - ECU communication
    - OBD protocol

closed 2018-02-24 by fr3ts0n