fr3ts0n/AndrOBD #42: Service 0x0A not supported

Vince-King, 2017-12-18:
Hi again

I installed my compiled version of AndroidOBD (no changes, and downloaded from the repo the other day).

I have one of these obd tools https://www.amazon.co.uk/gp/product/B074PQV4DY
and am using on my Mercedes c200k, 1.8, 2005

Every time I ask for OBD fault codes (permanent) I get the following error:
SEVERE	com.fr3ts0n.prot	(NRC:0x11) Service 0x0A not supported

I have now changed my logs to fine and will post a log after my next journey to the car.
Do you know what may be causing this?

Thanks

Vince

fr3ts0n, 2017-12-18:
NRC (Negative result code) 0x11 means: Service not supported

It means exactly what the message says:
Your vehicle does not support OBD service 0x0A "OBD fault codes (permanent)"

There is nothing wrong with your vehicle. 
I have not seen much cars yet, which do support this service. Mine doesn't either.
The only difference to my vehicle is: My vehicle just does not answer this service, where yours answers with a NRC (which triggers the dialog message)

Vince-King, 2017-12-18:
Thank you.
You say there is nothing wrong with my car.... did my log files get uploaded?

I have noticed a bug also... The app crashes out if you try to load a .txt file instead of the .obd file. Maybe a sanity check on the file type, to stop the app from crashing?

Thanks again

Vince

fr3ts0n, 2017-12-19:
What do you mean with log files?

If you mean the AndrOBD log files, yes the event is logged in AndrOBD.

If you mean the OBD history of the vehicle, No,  NRCs are not logged in the ECU. 
The only action OBD registers is if you clear fault codes.

Yes you are correct, a sanity check before loading measurement should help. Do you want to open another issue for this?

Thx

Vince-King, 2017-12-19:
Hi again

Quote "There is nothing wrong with your vehicle"
How did you know?
When I was out there testing my car.... Did my log files get uploaded to you on some master server? or just a lucky guess on your part?

Sure, Ill create a post re the sanity check :)

Vince

fr3ts0n, 2017-12-19:
That was just a comment on the car responding with a NRC message. There is nothing wrong with the car sending a NRC response.

No I haven't got any log messages. AndrOBD just stores log files for bug analysis on your mobile. You can even control how much information these log files contain ... (see: [Wiki](https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues#debug-log-files))

These log files don't get transferred anywhere as long as you don't send them to someone yourself. 
(You can verify this in the sources if you like)

closed 2018-01-11 by fr3ts0n