fr3ts0n/AndrOBD #45: Code clearing

MauroMombelli, 2018-01-13:
I tried to clean one code with the app, but the app give a warning it is going to delete a lot of stuff. Is that normal? is there a way to just reset one event?

fr3ts0n, 2018-01-13:
Yes, this warning message is normal.

Depending on the ECU this also may result in clearance of some learned values.
This usually is not a problem since the ECU can re-learn the values, but you may notice some strange engine behaviour while this re learning process is active.

No, you can not clear one single fault code if there are multiple faults set. 
You can only clear them all together.

closed 2018-01-15 by fr3ts0n

MauroMombelli, 2018-01-27:
sorry to bother you again, i see different app clearing the codes without giving this error.. is that just they hiding the stuff from the user?

fr3ts0n, 2018-01-27:
This message is not an error.

It is a information to the user that these things may be affected by the code clearing as well.
It should prevent a user from surprises, if the car for example starts to re-learn it's idle parameters, because they have been reset with the DFC clearance.

AndrOBD informs about these possible consequences of DFC clearance and asks for acceptance before   
any codes will be cleared. I the user does not want accept the possible consequences, then no code will be cleared.

If other apps clear DFCs without informing their users on possible consequences, it is up to them. 
I think it is just fair to the users to do it.

MauroMombelli, 2018-01-28:
i understand. can maybe the text be changed a bit to explain this is indeed the common and normal procedure?

MauroMombelli, 2018-01-28:
i used the english translation, i think it just should be added that this is the "normal" procedure.

When i saw that wall of text, i thoug i was running the wrong thing to remove my error code, and desisted to do it until i got clarification here.

fr3ts0n, 2018-02-03:
I have changed the dialog class of this message from ALERT to INFO.
This should indicate that this is a "normal" procedure.

Adding more text to a "wall of text" doesn't really make more people reading it.