fr3ts0n/AndrOBD #48: NRC: 0x22

Vince-King, 2018-01-19:
Hi again.

Im using the same version as my previous post, and am trying to clear an error code in a friends car (SL350 2010) and am getting NRC:0x22 Conditions not correct

Any idea why this is coming up, and how to make the conditions correct?

Thanks

Vince

fr3ts0n, 2018-01-19:
Hi Vince,

Is it possible that you try to clear fault codes while the engine is running, or the car is in motion?

This would be a classical scenario for "condition not correct" NRC.

Since conditions for this function  are specific to the vehicle / make i don't know them  all, but the ones above are the most common.

Vince-King, 2018-01-19:
yes it is possible, and I will try again tomorrow (had to much Jack Daniels tonight) thanks for the reply :)

Vince

closed 2018-02-03 by fr3ts0n