fr3ts0n/AndrOBD #50: Propose new logo

anaufalm, 2018-03-27:
Hi, I've seen your app and I think it's a lot of benefits, especially for the drivers. And here I would like to propose new logo for your app to have a good brand and become more interesting.

If you are interested and allowed me, I will make it for you, because I have an idea for this. ;)

fr3ts0n, 2018-04-02:
Hi @anaufalm ,

I already do have another offer to get a new logo created. ( #47 by @Ventto ) 
I would appreciate to get another suggestion for the artwork, but it may end up in something like a challenge between all of you.

I would like to suggest an election of the best logo/artwork when all drafts are available.

Would that be ok with you?

anaufalm, 2018-04-02:
Yeah sure, I have a draft but not ready yet. I'll present asap, soon maybe :)

fr3ts0n, 2018-04-10:
Hi @anaufalm ,
Thanks you for these design drafts. 
This is beautiful artwork.

My personal favorite currently is:
[image]

Do you see any chance to change the 2-lined version to read "androbd" rather than "andro obd"?
I understand that this would be much better for icons / etc. 

Thanks

anaufalm, 2018-04-10:
Yeah sure, go to fix it. So, you pick the purple?

anaufalm, 2018-04-10:
What kind size you want @fr3ts0n?

fr3ts0n, 2018-04-18:
Hi @anaufalm ,

Looks very nice to me. I also like the ones with the a and o being the wheels of the car ...

Do you already know which CC licence model you want to put this on?

This is also the last chance for @Ventto to challenge your design ...

anaufalm, 2018-04-18:
This is [yours](), all of them. Color, size, vector files, svg for scalable size, and pdf files. Also your icon, because this app is for android, so I make a size base on [Icon Design Guidelines](https://developer.android.com/guide/practices/ui_guidelines/icon_design.html), I created all. You can see in the folder. :)

fr3ts0n, 2018-04-18:
Thank you very much, CC BY is fine with me. You will definitely get the credits for this ...
I'll just leave @Ventto another week to challenge your design.
Is that ok with you?

anaufalm, 2018-04-19:
Yeah, its ok. Its up to you, how good what you think. I will be happy if my work is worn. :D

anaufalm, 2018-05-03:
Hi @fr3ts0n, how is it going? It's past two weeks. I seen @Venttoo didn't do it. 
What about my logo?

fr3ts0n, 2018-05-08:
Hi @anaufalm ,
Sorry for my late  response. I have been quite busy with my garden so I haven't been much online lately.

Yes, since @Ventto did not react on the possible challenge it will definitely be your design.
Please could you provide the corresponding files for icon, logo, svg for integration? 
(In a previous post you mentioned a folder which I did not find)

Thank you very much

anaufalm, 2018-05-08:
Sorry forget update the link, I will do it soon. Im on the way.

anaufalm, 2018-05-13:
Hi @fr3ts0n sorry Im late, [this](https://drive.google.com/open?id=1XP6E1lz6cFLE9Ynln2YchXlYu87lNqkL) is yours.

av2k, 2018-05-13:
@anaufalm @fr3ts0n  Thank you very much! The design looks really good. I'm not even a AndrOBD / ODB2 user but saw this ticket and it displays exactly the spirit why the open source scene is so good. 👍 :)

anaufalm, 2018-05-13:
@av2k Well, at first I found this app in F-Droid, and feel there is less with the logo. And that's the reason why I'm here. :D

fr3ts0n, 2018-05-31:
Integrated and released with V2.0.3 :smiley:

Your credits are linked to your GitHub page, I hope this is OK for you ...

[image]

Thank you again for your contribution.

closed 2018-05-31 by fr3ts0n

anaufalm, 2018-05-31:
Yeaay, its enough. ;)

anaufalm, 2018-06-01:
Bonus for you

[image]

[image]

fr3ts0n, 2018-06-01:
Hey, this is great for the start page ...
Thanks again

anaufalm, 2018-06-01:
You are welcome