fr3ts0n/AndrOBD #51: Failure to connect to a 2006 Toyota Sienna

gaponenko, 2018-04-02:
Hello,

When attempting to connect via bluetooth to an adapter plugged into a 2006 Toyota Sienna,  AndrOBD shows "ECU detect" in the status line, then goes "Offline".   I see this with v1.5.5 and v2.0.2.   Everything works fine when the same adapter is connected to a different, newer vehicle.   
ELM adaptive timing in the settings is "off".

Any ideas on how to tweak the settings to make it work?  A log file is attached.
[AndrOBD.log.1.txt](https://github.com/fr3ts0n/AndrOBD/files/1866846/AndrOBD.log.1.txt)

Andrei

fr3ts0n, 2018-04-02:
Hi @gaponenko ,

Thank you for your report. This is definitively a bug in the program.

Your vehicle seems to respond rather late to the initial request of it's supported features, where the ELM adapter responds with additional '.' characters in the meantime. 

AndrOBD erratically attempts to interpret these "." chars as a possible ECU address which leads to an error -> abort.

I'll fix that and let you know when it is available ...

fr3ts0n, 2018-04-02:
I've posted a test version for you on: [AndrOBD dev channel](https://t.me/AndrOBD_dev)

Please let me know if this fixes the problem ...

gaponenko, 2018-04-03:
Yes, the test version does work with the Sienna.  A log from a successful session is attached.
Thank you!
Andrei
[AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/1870174/AndrOBD.log.0.txt)

closed 2018-04-03 by fr3ts0n

takiainen, 2018-05-01:
I have the same issue when trying to connect with Nissan Almera N16 (year 2000). However, if I select from Settings -> OBD options -> preferred obd protocol -> ISO 9141-2, I can get connection, but usually I can view only one data menu before the connection is lost. The program just says "no data". Then I have to reconnect.

takiainen, 2018-05-01:
Could you link me the fixed version? I don't have telegram.

commit 5709db41 (2018-04-03): Fix #51 - ECU Address detection ISO9141 with preceeding ... chars