fr3ts0n/AndrOBD #52: Failed to connec to a 2017 VW Jetta

oroce, 2018-04-02:
Hey, I tried to connect to a 2017 (february) VW Jetta. After I select the bluetooth adapter the label goes online and immediately goes back to offline. the notification on the bottom says: `Device connection was lost.`

Torque lite and premium works without any problem.

In Torque I was able to connect to the car with turned off timing. As it can be seen in the log I tried with all the timing options.

[AndrOBD.log.1vwjetta.txt](https://github.com/fr3ts0n/AndrOBD/files/1868535/AndrOBD.log.1vwjetta.txt)

Android:
[image]

App version:
[image]

Fraifrai, 2018-04-03:
Hi,
same problem with SEAT IBIZA 2013

fr3ts0n, 2018-04-03:
Based on the log file of the VW Jetta it seems that your adaptor breaks the Bluetooth connection on the ATZ (reset) command. I have seen this happen with several adaptors already.

I would suggest to disable the ATZ command in the AndrOBD settings:
Settings -> Development options -> Disabled ELM commands -> check "ATZ"

This should eliminate the device reset and allow the bluetooth connection to stay alive.

Please let me know the results

Fraifrai, 2018-04-04:
This trick worked for my SEAT IBIZA ST FR 1.2 TSI 105

thanks

oroce, 2018-04-04:
@fr3ts0n thanks. Unfortunately, I don't have access to the Jetta right now, but I guess this has nothing to do with the car but rather with the adapter, right? So I should see the same issue with other cars and I guess that same trick would solve this problem.

Fraifrai, 2018-04-04:
@oroce : regarding my answer you should try. As it worked with my SEAT it shall work for any recent VW group car

fr3ts0n, 2018-04-04:
@oroce: Yes you are right. 
It is an issue of the combination of adapter- and mobile device. 
You will face the same situation on other cars as well, but the setting of **ATZ** command disabled should fix that situation for all vehicles.

Fraifrai, 2018-04-15:
@fr3ts0n it seems to be a recurrent problem. What about disabling this command in the default settings ?

fr3ts0n, 2018-04-18:
@Fraifrai This setting stays consistent with your installation for all following app usage.
This is only reset to default values if you de- and re-install the app, so I don't really see the necessity of changing the workaround into the default settings. 
The ATZ command is handled by most adapters without any problems in BT communication ...

Fraifrai, 2018-04-23:
maybe a comment about it in the settings should avoid such issues opening.

fr3ts0n, 2018-05-08:
I've added this subject to the [FAQ Wiki page](https://github.com/fr3ts0n/AndrOBD/wiki/Frequently-asked-questions)

closed 2018-05-08 by fr3ts0n

oroce, 2018-05-13:
I just wanted to confirm that disabling ATZ command resolved the issue for me, too. thanks @fr3ts0n