fr3ts0n/AndrOBD #53: Propagating an error up and reset a connection?

schnapi, 2018-04-10:
What would happen if we throw an exception here and propagate this error up to mainActivity. Can we do that? I would like to call mCommService.connectionLost() after that. Is that recommended?

https://github.com/fr3ts0n/AndrOBD/blob/eceacfc63ecb96781c75f6a39d52e0e1693ee0be/library/src/main/java/com/fr3ts0n/prot/StreamHandler.java#L103

Thanks!

fr3ts0n, 2018-04-10:
Hi @schnapi ,

Yes, I can see the advantage of it. 
I also see that the WifiHandler does exactly behave like you suggested, but BT and USB do not.

There are quite  lot of callers to this method, so in order to propagate the exception I will have to check each caller to handle the exception.

Please give me some time for analysis ...

Thanks

fr3ts0n, 2018-04-18:
Currently doesn't look too well ...
I tried to do these changes but received several un-intended effects during my real-live tests (like stopped communication w/o any disconnect detection ...). 
I'll have to find a better way to clean up this implementation ... 

If you have a good idea, pls suggest with a pull request ...

schnapi, 2018-04-26:
So far I haven't notice any un-intended effects from my side. Maybe you have problems with OBD adapter. I had some cheap OBD adapter which caused me a lot of problems. If I have any problems I will try to find another solution...