fr3ts0n/AndrOBD #54: Can the collected data also include GPS?

itamar-otonomo, 2018-04-22:
That way I would be able to correlate OBD data with location and speed?

fr3ts0n, 2018-05-08:
That sounds like a good AndrOBD extension plugin project ...

see: https://github.com/fr3ts0n/AndrOBD-Plugin

Anybody interested to create this?

jgpATs2w, 2018-07-07:
Hi, I was planning to include the functionality to send data to a remote server, and already have an android project with gps data collection, so I would do it!
I've made a question about the creation of a new plugin: https://github.com/fr3ts0n/AndrOBD-Plugin/issues/3

abalage, 2019-06-01:
I would love to see this feature. I can offer my help for testing.

Currently I use 'Carscanner' to have CSV export and Trackbook (https://f-droid.org/en/packages/org.y20k.trackbook/) to have GPS data and merge them together with a script. It is uncomfortable and prone to errors (timing mismatch).

fr3ts0n, 2019-08-25:
Test versions are available now for your tests on [AndrOBD dev channel](https://t.me/AndrOBD_dev)
- AndrOBD V2.1.0 (for support of handling plugin provided data in data list)
- GpsProvider plugin V1.0.0

fr3ts0n, 2019-08-25:
Support added with V2.1.0 + GpsProvider V1.0.0

closed 2019-08-25 by fr3ts0n

abalage, 2019-08-25:
>
> Test versions are available now for your tests on [AndrOBD dev channel](https://t.me/AndrOBD_dev)
> 
Unfortunately meanwhile I have already purchased another tool for the same purpose.
However I really appreciate the effort you did put into this feature.