fr3ts0n/AndrOBD #55: VIN decoding

schnapi, 2018-04-24:
https://github.com/fr3ts0n/AndrOBD/blob/eceacfc63ecb96781c75f6a39d52e0e1693ee0be/library/src/main/java/com/fr3ts0n/ecu/prot/obd/ObdProt.java#L600-L603

Example of vin response from https://www.elmelectronics.com/wp-content/uploads/2016/07/ELM327DS.pdf:
                            handleTelegram("013".toCharArray());
                            handleTelegram("1:4902015956".toCharArray());
                            handleTelegram("2:314D5336363432".toCharArray());
                            handleTelegram("3:3632313730353034".toCharArray());

I think you have to remove also 3 byte (01).
The third byte (the '01 '), tells the number of data items that are to follow (the vehicle can only have one VIN, and this agrees with that)

e.g.
```
String payLoad = String.valueOf(getPayLoad(buffer));
dataItems.updateDataItems(msgService,
msgPid,
hexToBytes(payLoad.substring(2)));
```

fr3ts0n, 2018-05-08:
Thank you for your detailed report.
You are correct, this byte is already interpretet as VIN content.

Is this 3rd byte mandatory or optional?
Do you have any document which describes this byte?

If it is mandatory with all 0902 answers then I can change the start offset within the data description of the VIN response.

fr3ts0n, 2018-05-10:
OK, I found it in the OBD docs. It's a mandatory byte.
I'll fix it in the data descriptions ...