fr3ts0n/AndrOBD #57: WARNING stream RX error

JonnyTech, 2018-05-19:
When connecting it automatically disconnects with log showing `WARNING stream RX error`: 
[AndrOBD.log.1.txt](https://github.com/fr3ts0n/AndrOBD/files/2018994/AndrOBD.log.1.txt)

Using https://github.com/pires/android-obd-reader works correctly: 
[Log_18_05_2018_13_03_01.csv.txt](https://github.com/fr3ts0n/AndrOBD/files/2018998/Log_18_05_2018_13_03_01.csv.txt)

Is there anything that I can try?

fr3ts0n, 2018-05-20:
Hi @JonnyTech,
This appears to be the same issue as #52 but with a different effect in combination with your mobile:

There seems to be a new batch of cheap ELM clone adapters which break the bluetooth connection on rception of a ATZ (reset) command.
see also in the [FAQ section of wiki:](https://github.com/fr3ts0n/AndrOBD/wiki/Frequently-asked-questions#bluetooth-disconnected-right-after-initialisation--connection-attempt)

The workaround for this is to disable the ATZ command in the AndrOBD settings:
Settings -> Development options -> Disabled ELM commands -> check "ATZ"

This should eliminate the device reset and allow the bluetooth connection to stay alive.

Please let me know the results

JonnyTech, 2018-05-20:
Thanks @fr3ts0n, that worked - although it took a few attempts to connect but maybe it was because the adaptor was in the process of resetting?

Once connected I got a blank white screen. Another bug. White text on white background. Switching to night mode made the text show up.

Shall have a play tomorrow and provide further feedback.

closed 2018-11-03 by fr3ts0n