fr3ts0n/AndrOBD #60: App not connecting to ECU

bollejansson, 2018-09-02:
Using the current version(s) of AndrOBD I cannot connect to my car's ECU (2009 Fiat Grande Punto, 1.4l Petrol Engine; 500 kBit, 29 Bit ID protocol) using my ELM327 bluetooth dongle ("China").

The last version known to work for me is 1.4.7, but surprisingly even then only when deliberately choosing one wrong protocol: (ISO 14230-4 KWP (5 Baud Init)). Even then it flashes "DISCONNECTED" before turning to "CONNECTED". If I understand the attached log file correctly, the App tries connecting using "Automatic" as protocol after using "wrong" protocol failed.

Choosing "Automatic" as Protocol in first place does not work either, neither does, as I stated, choosing the correct protocol stated in the first sentence.

I tried using other OBD apps choosing the correct 500 KBit/29 Bit protocol, which could connect to the ECU without problems.

I did not cut down the log files as I didn't want to remove important information from them. If you need more information I'll be gladly providing it for you.
[old_version.log](https://github.com/fr3ts0n/AndrOBD/files/2343497/old_version.log)
[new_version.log](https://github.com/fr3ts0n/AndrOBD/files/2343496/new_version.log)

bollejansson, 2018-09-06:
Thanks for your quick reply!

I managed to record the android system log while running the app and trying to connect to the ECU. Starting from line 137 (09-06 21:05:57.554) there's an RX error followed by a NumberFormatException. I assume this might be a useful hint?

[logcat_output.txt](https://github.com/fr3ts0n/AndrOBD/files/2358477/logcat_output.txt)

fr3ts0n, 2018-09-07:
Here is a test version for you, which includes handling of 29 bit CAN ID's for ECU selection.
https://t.me/AndrOBD_dev/208

Please check and report if this works for you

bollejansson, 2018-09-07:
this version works like a charm!
thank you very much :)

commit aa77271f (2018-09-08): Fix #60, #61 - improve ELM message handling

#60 Handle 29 bit CAN IDs for ECU detection
#61 Ignore verbose broadcasts from ELM clone
closed 2018-09-08 by fr3ts0n