fr3ts0n/AndrOBD #61: can't connect elm327

RookieBirds, 2018-09-05:
Hello, I read the related question, disable the ATZ command, but still connect->offline when I click the connect button. 
[AndrOBD.log.3.txt](https://github.com/fr3ts0n/AndrOBD/files/2352938/AndrOBD.log.3.txt)

fr3ts0n, 2018-09-05:
Thanks for your report.

What kind of adapter is this?
Please could you provide a link and/or specification?

Based on the Bluetooth ID and broadcasted connection message this does not look like a ELM327 at all.

> CommService connect to: AA:BB:CC:11:22:33
> ...
> ELM rx:'+CONNECTING<<94:65:2D:9E:DF:B5'

RookieBirds, 2018-09-06:
thank you for your reply.
This is a Chinese product, bought at Alibaba's Taobao online store, you can visit this link to view its description---https://item.taobao.com/item.htm?spm=a230r.1.14.60.529b48d564m5az&id=521701769759&ns=1&abbucket=19#detail. My tutor and I used the Torque app to find out which ones can be connected. The OBD protocol for the Torque display connection is ISO15765-4CAN11/500. Do I need to change a new ELM327 to suit your app, and if so, where to buy it.
I don't know if my description gives you some information and I look forward to your answer!

fr3ts0n, 2018-09-07:
Here is a test version which should ignore such verbose messages like the one above.
https://t.me/AndrOBD_dev/208

Please check and report if this works for you