fr3ts0n/AndrOBD #62: Error loading saved *.obd data

preet, 2018-09-09:
I receive the message 'invalid stream header' when trying to load saved *.obd data. I'm using AndrOBD version 2.0.3. Here is the logcat stacktrace:

```
09-09 17:52:41.562  1595  3577 E Load measurement: java.io.StreamCorruptedException: invalid stream header: 89504E47
09-09 17:52:41.562  1595  3577 E Load measurement:      at java.io.ObjectInputStream.readStreamHeader(ObjectInputStream.java:805)
09-09 17:52:41.562  1595  3577 E Load measurement:      at java.io.ObjectInputStream.<init>(ObjectInputStream.java:302)
09-09 17:52:41.562  1595  3577 E Load measurement:      at com.fr3ts0n.ecu.gui.androbd.FileHelper.loadData(FileHelper.java:210)
09-09 17:52:41.562  1595  3577 E Load measurement:      at com.fr3ts0n.ecu.gui.androbd.FileHelper$2.run(FileHelper.java:185)
```

I have three separate obd files, all different sizes and I receive the same error for all of them. My exact process to open the files were to start up AndrOBD, select the '...' icon in the top right and select 'Load measurement'. I've attached one of the files.
[2018.09.03-14.43.22.obd.zip](https://github.com/fr3ts0n/AndrOBD/files/2364710/2018.09.03-14.43.22.obd.zip)

Is this format a custom binary or something I can try opening with another util?

fr3ts0n, 2018-09-11:
Thank you for your report,
The format of .obd files is a set of serialized java objects.
It can be only loaded with AndrOBD again, but also be exported to CSV data from Graph display.
 
Did you save the provided file with the same version of AndrOBD? 
It appears to contain a different version (SerialVersionUid) of measurement history objects.

preet, 2018-09-11:
I am pretty sure I used the same version. I've only installed AndrOBD once (from F-droid) and I have automatic updates disabled. I will try to save another set of measurements and update with my results... in the mean time, if I understood you correctly is there a way to open the data files with another version of AndrOBD? A couple of the files were from longer trips and I'd like to access the data if possible.

fr3ts0n, 2018-09-12:
I have tried to load your provided .obd file on:
* Android 5 : Similar error message as you reported
* Android 7 : OK
* Android 8.1 (Virtual device) : OK
I assume that the serialisation format differs between different Android versions.

Have you saved the file with the same device & version of Android which shows the error?
What version/flavour of Android do you use?

preet, 2018-09-29:
Hi, sorry for the delay; I lost my OBD connector until it turned up again today. For some reason I can no longer recreate this problem. I'm on the same AndrOBD version (2.0.3), and have not updated my device outside of updating F-Droid (though I don't think AndrOBD was updated here).

I created another set of data, saved it and then immediately opened it and it worked fine. Then I tried to open the data I uploaded here as well, and it also worked. I tried this a couple of times, closing AndrOBD between each attempt and no issues.

Just for reference, some device info:
* Nexus 4
* Android version: 7.1.2
* LineageOS version: 4.1-20180201-NIGHTLY-mako
* LineageOS API Level: Guava (7)

Should I close this issue? I can re-open it if the problem returns/I can recreate it.

closed 2018-10-21 by fr3ts0n