fr3ts0n/AndrOBD #67: Unable to connect bluetooth ELM1.5

mulx, 2018-11-01:
Hello,
I'm trying to connect to my car with my bluetooth ELM but every time I hit the connect button I get a connection failed.

Other apps, like torque, Car scanner, EODB Easy or ScanMaster works fine out of the box with the same car & device).

I've tried disabling ATZ as suggested in the FAQ but that change nothing. 

My car is a Renault Clio 3, the protocols others apps use is ISO 15765-4 CAN 11 bits ID 500kbits, I've also tried (even thought the last test are in automatic mode).

I've extract the logcat during the time I tried to connect, hope it might help.

Regards,

[logcat_extract.txt](https://github.com/fr3ts0n/AndrOBD/files/2538745/logcat_extract.txt)

mulx, 2018-11-01:
Also this apps used to connect fine: https://github.com/pires/android-obd-reader/ (even if it's not developed anymore, didn't check recently). 
Also, I'm using Android LineageOS 15.1 (Android 8.1).

fr3ts0n, 2018-11-03:
Thanks for reporting this issue.

After searching around I found a elm 1.5 adapter which behaves the same way.
I compared the source of BTCommService with the project you mentioned and found something:
The BT handler in above project creates a BT fallback socket, if the initial connect() on the regular SPP socket fails. 
I havent got the details what the fallback socket does different, but it appears to work for my adapter showing the problem. 
I suspect the difference is a matter of BT port selection, but that is only a suspicion.

Please find here a test version for your tests (and hopefully confirmation of the fix).
https://t.me/AndrOBD_dev/210

fr3ts0n, 2019-02-02:
Fix released with V2.0.5

closed 2019-02-02 by fr3ts0n