fr3ts0n/AndrOBD #73: Add support for tasker

krzyk, 2019-04-05:
It would be good to be able to:
1. Connect to OBD
2. Start collecting data
3. Publish all data using MQTT
4. Disconnect from OBD

Using Android Intents, and publish them on wiki so one can call them from tasker.

My use case is to have publish all OBD data collected published when my phone connects to my car Bluetooth (send intent to connect to OBD/start collecting data/publish data to MQTT), and stop it when my phone disconnects from bluetooth.

Example of great tasker integration is GPSLogger project, see their FAQ regarding Tasker here: https://gpslogger.app/#controllinggpslogger

gabriellluz, 2019-05-14:
I second that.

Tasker misses some solution for OBD. We can automate anything but we can't acces OBD data. I really think that would be the first app of this kind with Tasker support.

fr3ts0n, 2019-05-18:
Sorry for not responding so long.
I am currently working on another project, so I haven't looked into the details yet.

Couldn't this be done by the AndrOBDMqttPlugin just by a automated start of AndrOBD (by Tasker) on BT connect?

All the rest could be done by the AndrOBD startup settings.

fr3ts0n, 2023-07-16:
The only thing to be automated for your scenario is the automated start/stop of AndrOBD on BT connect/disconnect.

All internal settings of AndrOBD startup behaviour can be set in AndrOBD preferences.
(see: https://github.com/fr3ts0n/AndrOBD/wiki/Automation-on-startup)

wunter8, 2023-07-28:
I just tried it again today. I have all the checkboxes checked under "common settings" > "use last settings", but when I open the app, nothing happens until I click the "connect"/plug-shaped icon. Is that how it's supposed to be?

fr3ts0n, 2023-08-06:
You are correct. The app currently requires a manual "connect" action. 
I'll fix that so it will auto-connect when started and auto-connect is preferred

wunter8, 2023-08-06:
That'd be awesome! Thanks!

closed 2023-08-06 by fr3ts0n

wunter8, 2023-08-06:
That was quick. Thanks! I'll watch for the update through F-Droid

fr3ts0n, 2023-08-06:
Test version is available here:
https://t.me/AndrOBD_dev/252

F-Droid update will be triggered soon

commit 0b852e7d (2023-08-06): Fix #73 - Auto connect at startup on preference