fr3ts0n/AndrOBD #74: Bug found in AndrOBD app Bitcoin option is not working properly

facebook-786, 2019-05-18:
# Project Information

* Repository: https://github.com/fr3ts0n/AndrOBD
* Project Name: AndrOBD

# Expected behavior

When I click on the bitcoin option, then an address of bitcoin should be displayed where users can donate to dollars.

# Actual behavior

An error appears when I click on the bitcoin option

# How to reproduce

1-[Download ](https://f-droid.org/packages/com.fr3ts0n.ecu.gui.androbd/)the app and open it
2- Click the more option at the top right
3-Click on the setting
4- Click on AndroOBD
5- Click on the bitcoin option
6- see bug 

* Browser/App version: v2.0.7
* Operating system: 6.0.1
* Smartphone: Galaxy j7 

# Recording Of The Bug

[image]

# My Github Account
https://github.com/facebook-786

facebook-786, 2019-05-20:
hello sir @fr3ts0n please review my contribution

fr3ts0n, 2019-05-21:
Do you have any App installed which van handle BitCoin transactions?

facebook-786, 2019-05-21:
Sir, I do not have any such application available

fr3ts0n, 2019-05-21:
That's the reason for the Error Message. In Order to do bitcoin donations you need a bitcoin app installed. AndrOBD      can't handle this by itself.

facebook-786, 2019-05-21:
Already have a Bitcoin application installed on my mobile phone. This error is in your application

fr3ts0n, 2019-05-23:
Re-checked this with **Bitcoin Wallet** from FDroid. The app opens immediately as expected. 
Which app do you use?
Maybe this app does not handle bitcoin: URLs.

facebook-786, 2019-05-23:
[image]

after downloaded a bitcoin wallet version 1.2.0
application crash when I click on bitcoin address you can see in bug GIF easily

fr3ts0n, 2019-05-24:
AndrOBD does request for a bitcoin activity by requesting the viewer of a bitcoin URL.
The format of the requested URL does match the specified bitcoin url scheme. @see https://bitcoin.stackexchange.com/questions/4987/bitcoin-url-scheme

If the installed BitCoin app aborts with an error when handling this (correctly formatted) URL-request, I would suggest to get the BitCoin app fixed rather than AndrOBD, since it doesn't seem to follow the specifications.

In case there is no BitCoin app installed, I would consider the indicated error message from AndrOBD to be reasonable descriptive to identify that there is no app to handle the bitcoin request.

closed 2019-05-24 by fr3ts0n