fr3ts0n/AndrOBD #75: OBD Protokollfehler "(NRC: 0x78) Request correctly received but response is pending"

74sb, 2019-06-10:
**OBD Protokollfehler "(NRC: 0x78) Request correctly received but response is pending"**
After this error message no (other) car information is availabe.

Steps to reproduce the behavior:
1. Start App and wait for connection
2. Click on "OBD Fahrzeug Info" 

Expected behavior:
If the error is based on hardware (car) behavior, it should not block / getting other information. 

**Important details on my environment:**
- Android OS version: 9
- AndrOBD version: 2.0.7
- OBD adapter type: tinxi ELM327 Interface (OBD II WiFi tinxi.com GmbH), WiFi Connection used
- Connected vehicle: Ford Focus, 2011, 1.6 TDCi 80kW

** logs **
[farzeuginfo-AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/3273187/farzeuginfo-AndrOBD.log.0.txt)

fr3ts0n, 2019-06-16:
I can see in the log file, tat AndrOBD is reacting with an ATZ (Reset) command to the reception of an NRC response.
Do you have the OBD config option "Reset on NRC" activated?

Please could you re-try with this option de-activated (which is the default setting)? 
Then it should be possible to receive the vehicle response correctly.

74sb, 2019-06-18:
Yes config option "Reset on NRC" is activated. I tried it first without. Got same message. That's why i activated it. I will send you logs with "Reset on NRC" deactivated. But this can take a few days because i have no access to the car.

fr3ts0n, 2019-06-23:
I will close this issue and follow this with the enhancement of #76

closed 2019-06-23 by fr3ts0n