fr3ts0n/AndrOBD #76: OBD Protokollfehler "(NRC:0x11) Service 0x0A not supported"

74sb, 2019-06-10:
**OBD Protokollfehler "(NRC:0x11) Service 0x0A not supported"**
It seems to me that error message describes a unsupported car feature.
Another sample message is "(NRC:0x12) Sub-Function not supported (SVC:0x10) 

Steps to reproduce the behavior:

    Start App and wait for connection
    Click on "OBD Fehlercodes (permanent)"

Expected behavior:
If the error is based on a unsupported feature response, the results should visualized in a normal window. At least for NRC:0x11 and NRC:0x12, for a better user experience :).

If it is too hard to change the program a wiki page which describe some common infos/warnings like  NRC:0x22 (#48), NRC:0x11 (#42), NRC:0x12 would do it.

Important details on my environment:

    Android OS version: 9
    AndrOBD version: 2.0.7
    OBD adapter type: tinxi ELM327 Interface (OBD II WiFi tinxi.com GmbH), WiFi Connection used
    Connected vehicle: Ford Focus, 2011, 1.6 TDCi 80kW

** logs **
[fehlercodespermanent-AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/3273413/fehlercodespermanent-AndrOBD.log.0.txt)

fr3ts0n, 2019-06-16:
It appears that your vehicle doesn't support the service of permanent DFCs.
I can see in the log file, tat AndrOBD is reacting with an ATZ (Reset) command to the reception of a NRC response. After the Reset the Reset, the adapter does just respond with prompts, but not with any status message.

Do you have the OBD config option "Reset on NRC" activated?

Please could you re-try with this option de-activated (which is the default setting)? 
This should keep communication going ...

On the visualisation:
Depending on the error reaction to a NRC response (Reset/NON Reset) the message shall not need any confirmation.

74sb, 2019-06-18:
Yes. I will re-try it.

fr3ts0n, 2019-06-23:
I have looked deeper into th handling of NRC's:

The current implementation doesn't use any classification between the NRC responses, but this is obviously missing to determine different reactions upon their response.
i.e.: a "Response pending" NRC has to be handled differently than a "Service not supported", since there is still a valid response to be expected in the first case. (Currently the used service is closed on any NRC response)

I will enhance the NRC handling to allow specific reactions on each NRC.

If you are interested and have any suggestions on the expected reactions, here is the list of NFC's:

Hex | Description | Exp. Reaction
-- | -- | --
10 | General reject |  
11 | Service not supported |  
12 | Sub-Function not supported |  
13 | Incorrect message length or invalid format |  
14 | Response too long |  
21 | Busy repeat request |  
22 | Conditions not correct |  
24 | Request sequence error |  
25 | No response from sub-net component |  
26 | Failure prevents execution of requested action |  
31 | Request out of range |  
33 | Security access denied |  
35 | Invalid key |  
36 | Exceeded number of attempts |  
37 | Required time delay not expired |  
38-4F | Reserved by Extended Data Link Security Document |  
70 | Upload/Download not accepted |  
71 | Transfer data suspended |  
72 | General programming failure |  
73 | Wrong Block Sequence Counter |  
78 | Request correctly received  but response is pending |  
7E | Sub-Function not supported in active session |  
7F | Service not supported in active session |

fr3ts0n, 2019-06-23:
Here are my proposals for classification of the NRC's:
Please let me know if this makes sense ...

## Display classes
Memonic | Action
-- | --
ERR | Error dialog (with confirmation)
NOTE | TOAST note (w/o confirmation)

## Reactions
Memonic | Action
-- | --
IGN | Ignore message, wait for valid response
SKIP | Skip message, continue next service request
REPEAT | Repeat current service request
SVC0 | Close current service
RST | Reset Adapter (if allowed by config)

## NRC responses / Classifications
Hex | Mnemonic | Description | Display | Reaction | Notes
-- | -- | -- | -- | -- | --
10 | GR | General reject | ERR | RST |  
11 | SNS | Service not supported | ERR | SVC0 |  
12 | SFNS | Sub-Function not supported | NOTE | SKIP |  
13 | IMLOIF | Incorrect message length or invalid format | NOTE | SKIP |  
14 | RTL | Response too long | NOTE | SKIP |  
21 | BRR | Busy repeat request | NOTE | REPEAT |  
22 | CNC | Conditions not correct | ERR | SKIP |  
24 | RSE | Request sequence error | ERR | SVC0 |  
25 | NRFSC | No response from sub-net component | ERR | RST |  
26 | FPEORA | Failure prevents execution of requested action | ERR | SVC0 |  
31 | ROOR | Request out of range | NOTE | SKIP | Assumption: Incorrect sub-service / pid
33 | SAD | Security access denied | ERR | SVC0 |  
35 | IK | Invalid key | ERR | RST |  
36 | ENOA | Exceeded number of attempts | ERR | RST |  
37 | RTDNE | Required time delay not expired | NOTE | REPEAT | Time delay specified? (how long to wait?)
38-4F | RBEDLSD | Reserved by Extended Data Link Security Document | ERR | SVC0 |  
70 | UDNA | Upload/Download not accepted | ERR | SVC0 |  
71 | TDS | Transfer data suspended | NOTE | REPEAT |  
72 | GPF | General programming failure | ERR | SVC0 |  
73 | WBSC | Wrong Block Sequence Counter | ERR | SVC0 |  
78 | RCRRP | Request correctly received  but response is pending | NOTE | IGN |  
7E | SFNSIAS | Sub-Function not supported in active session | NOTE | SKIP |  
7F | SNSIAS | Service not supported in active session | ERR | SVC0 |

fr3ts0n, 2019-06-23:
I do have a test version ready for you, which handles each NRC towards the table above 
(for display and protocol reaction).
https://t.me/AndrOBD_dev/216

Pls let me know if this solves the issue for you.