fr3ts0n/AndrOBD #77: How to add service functions for land rover defender

meoux23, 2019-06-15:
Could you please take some time out to guide me about adding a new tab in the drop down menu. Land rover has special testing functions that require a set of ecu commands. If I could set up one test only, it would complete my college project which is by far stuck :( 
I'd be really really thankful.

fr3ts0n, 2019-06-16:
If you are mentioning additional UDS services to be supported, it is pretty much all OEMs offer some of these...
The problem with these serices is: They are not standardized as OBD services are. 
It needs a lot of research and studying the OEMs documentation to get the right data ID's and conversions for the corresponding vehicle. 
These data items are definitely specific to a vehicle OEM and may be even specific to certain vehicles only.

The amount of work required for research and implementation does not really justify for development of a free & open sourced app.

fr3ts0n, 2019-06-23:
## What I would do to start with:
### Define your project
* Read through the [AndrOBD wiki](https://github.com/fr3ts0n/AndrOBD/wiki)
  - Especially the parts: 
    - [data model](https://github.com/fr3ts0n/AndrOBD/wiki/Configuration-data-model)
    - [data customization](https://github.com/fr3ts0n/AndrOBD/wiki/Data-customization)
* Read Rover OEM diagnostic documentation
  - define service(s) and PID('s) you want to support

### Proposed code changes
* Define your new service / PID data
  - Add service/pid data into pid.csv
  - If additional conversions are required, enter these in conversions.csv
* Create a new enhancement class from (ObdProt / ElmProt) i.e.: RoverProt
  - Override setService and handleTelegram messages to handle new service messages
* Create a new enhancement class to the main GUI (MainActivity) i.e: RoverActivity
  - add menu item for additional service
  - override menu hander for new MenuItem selection
* Make App start from new, enhanced GUI class
  - Adapt AndroidManifest

### Test your code
* create test instances for your service and all eventual failure cases (they will occur ... ;-) )
  - run all the tests

### Povide changes
* After your functions work, please provide the changes via pull request
  - Since AndrOBD is GPL V3+ license