fr3ts0n/AndrOBD #81: (maybe RFE) read and store accelerometer

atrent, 2019-07-25:
Hi, is it possible to read and store (in the exported CSV) phone accelerometer values?
Thank you

fr3ts0n, 2019-07-28:
Currently AndrOBD only records OBD based values.
There is another, similar RFE #54 for integration of GPS values

If anybody is interested in contributing a new plugin for integration of accellerometer, and/or GPS data, every help is welcome.

atrent, 2019-07-28:
I didn't notice RFE #54, thanks, I'm closing this one

fr3ts0n, 2019-08-25:
Test versions are available now for your tests on [AndrOBD dev channel](https://t.me/AndrOBD_dev)
- AndrOBD V2.1.0 (for support of handling plugin provided data in data list)
- SensorProvider plugin V1.0.0

fr3ts0n, 2019-08-25:
Support added with V2.1.0

closed 2019-08-25 by fr3ts0n