fr3ts0n/AndrOBD #82: Fuel consuption not showing up

alex9099, 2019-07-31:
Hi, 

I just received a odb2 adapter for my car but the app does not show the fuel consumption, it seems to use the 0x01 0x5e code to get the fuel consumption (this if i understood correctly), but it does not show up on the list, does this mean that my car does not support reporting the fuel consumption? it does an average fuel consumption on the dashboard, but it would be nice to get a realtime fuel consumption gauge.

Does the app try everything from the pids.csv file?

fr3ts0n, 2019-08-01:
The App tries everything the vehicle supports. OBD protocol indicates bit masked which PIDs are supported by the vehicle.
Pls could you provide a Log file for me to verify correct function?
Thx

fr3ts0n, 2019-08-01:
Another possibility is the pid/data item is masked out for display, but this also can be verified in the log.

alex9099, 2019-08-01:
Where can I see the log file? Would "info" logging be enough?

fr3ts0n, 2019-08-01:
Pls see for location of log files:
https://github.com/fr3ts0n/AndrOBD/wiki/Reporting-improvements---issues 

Log level **FINE** would be helpful, since this will contain all communication.
Thx

alex9099, 2019-08-01:
Thanks, I'll give it a look later today. 

Can you improve a little bit the documentation on the pids/conversion template files? I don't quite understand what each value does (there are some other pids for fuel consumption that I would like to try, for some reason I can't find them, but I am almost sure I saw them)

alex9099, 2019-08-09:
I totally forgot about this. well, here goes the log ;) i dont see anything about fuel consumption
[AndrOBD.log.0.txt](https://github.com/fr3ts0n/AndrOBD/files/3488157/AndrOBD.log.0.txt)

fr3ts0n, 2019-08-25:
Yes, you are correct. 
Your car does not seem to support any PID about fuel consumption. Sorry about that.

However, in your logs I can see, that PID 41 does not get resolved by AndrOBD at all. 
This PID provides bit-masked status information only, but I will add this, so the data list is complete ...

alex9099, 2019-08-25:
That seems to be for some sensor tests right?

About the tests on the top (fuel system and so on), how would i trigger them?

So in this car there would be absolutely no way to calculate the fuel consumption? I saw something about the input air pressure or something like that, which could be used for an aproximated fuel consumption reading, but i'm not sure how that would work

fr3ts0n, 2019-08-25:
Yes, it's about system tests (looks very similar to PID 1)

These tests are automatically triggered and performed by your vehicle during a regular drivecycle. They are not triggered from the outside.

On fuel consumtion I am pretty sure there are data items with the OEM's specific diagnostic protocol, but obviously not supported on OBD level. I am afraid there is nothing AndrOBD can do about that.

closed 2019-08-25 by fr3ts0n