fr3ts0n/AndrOBD #84: Feature request: Add to Play Store

user8446, 2019-09-05:
Hello,
AndrOBD is a great a lightweight app. Publishing on the Play Store would expose it to users that aren't familiar with side loading or alternate app stores such as FDroid. Are there reasons that it isn't published there?
Thank you for the development!

fr3ts0n, 2019-09-21:
Hi @user8446 
You are correct,this would broaden the market for AndrOBD considerably.
- Since the clients would be a) more, b) with less technical knowledge, this would also mean to increase the 1st level support drastically.
- On any structural change of the project, the overhead effort for release preparation, is quite something to be considered. This is per release platform. 
- I like to support open source projects, which F-Droid is, and supports.

Since I am operating this as a spare time project, I think it is currently is enough to publish on F-Droid only.

user8446, 2019-09-21:
That makes perfect sense. Thank you again!

closed 2019-10-04 by fr3ts0n